ilg-archived/qemu #2: gnuarmeclipse/images is not installed, when QEMU is built from sources

RomanSaveljev, 2015-09-14:
Hello, I have tried to build gnuarmeclipse QEMU from sources in Ubuntu:

* QEMU version: qemu-gae-2.3.50-20150805
* System version: Ubuntu 14.04

To build:

```shell
$ ./configure --target-list=gnuarmeclipse-softmmu --prefix=/opt/gnuarmeclipse-softmmu
$ make -j4
$ sudo make install
```

What I found out is that `gnuarmeclipse/images` is not copied under `/opt/gnuarmeclipse-softmmu/share/qemu/`, which further stops the emulator from launching.

As a workaround I have copied the images myself.

ilg-ul, 2015-09-14:
Please check the QEMU build script, available separately in the build-scripts git.

It does quite a lot of things, including adding the images to the package.

The classical configure+make+install is no longer supported, but if you think useful to maintain it, and provide patches, I can apply them.

ilg-ul, 2015-09-14:
BTW, did you try the provided [binaries](https://github.com/gnuarmeclipse/qemu/releases)? do they work on your machine?

RomanSaveljev, 2015-09-14:
Thanks for quick reply. I am on Linux and there are only Windows binaries? I read the build-scripts and it seems quite complicated.. I have to study it more.

I am not sure I can fix the classical way to embrace everything the new build script does.

ilg-ul, 2015-09-14:
> only Windows binaries

nope. GitHub Releases are a bit confusing, but binaries for all platforms are available there.

https://github.com/gnuarmeclipse/qemu/releases/tag/gae-2.3.50-20150816

ilg-ul, 2015-09-14:
> I am not sure I can fix the classical way to embrace everything the new build script does.

no need for this, just what you consider necessary.

closed 2016-04-04 by ilg-ul