ilg-archived/qemu #3: Floating point?

mlyle, 2015-09-25:
Hey everyone.  I see floating point on Cortex M3/M4 isn't supported.  Apologies because I don't really know QEMU internals that well.

I know there's a number of things that need to be done infrastructure-wise to fully correctly support floating point.  Probably most difficult is the need to delay storing FP state in interrupt handlers until a floating point instruction occurs, which it doesn't look like there's a good way to do in QEMU.

It seems we also would need to limit to single precision.

But could an early approach just be to unconditionally store floating point state in the lazy mode?  I understand it's not absolutely correct and bad for performance, but pretty much all existing code would work.  I currently am working on TauLabs, an open-source flight controller for multicopters, planes, and other flying vehicles, and we'd really like to get automated integration test together.  However, on STM32F3 and STM32F4 platforms we heavily use hardware floating point.  Because of upstream libraries we use, it's challenging to build with just soft-float.

ilg-ul, 2015-09-25:
That's correct, storing FP state in interrupt handlers is the main thing to be done in order to implement FP in Cortex-M[347]. It can be done, but it'll take some effort, and I currently do not have the resources to do it.

If you are willing to contribute patches, I can consider them; also if you know someone interested to sponsor QEMU development, I can reconsider priorities.

mlyle, 2015-09-25:
Would you look favorably upon patches that do everything except the lazy portion?  e.g. store the state upfront only (in lazy or unconditional modes) or never store the state?

ilg-ul, 2015-09-25:
From what I could understand the QEMU maintainers strict requirements, I doubt they will accept non-standard implementations, and I concur to this position.

mlyle, 2015-09-25:
OK.  I won't bother then.

ilg-ul, 2016-04-04:
I plan to fix some other issues in qemu, and possibly address the M4 FPU issues. Are you still interested to contribute to the project?

qrthur, 2016-08-30:
Any news/deadline/beta regarding the M4 FPU support?

I wish I could help, but I am not experienced enough in this.

ilg-ul, 2016-08-30:
unfortunately no clear date set for it.

qrthur, 2016-08-30:
Alright, I will stay tuned then :-)
Thanks!

ilg-ul, 2016-08-30:
for those interested, I repeat some details: the FPU code is already there. what is missing is the correct exception processing to do the lazy mode.

unfortunately the exception processing code in QEMU is not properly implemented, the NVIC is piggy backed on top of some larger ARM interrupt controller (GIC), and the implementation is far from appropriate.

I plan to reimplement NVIC from scratch, to remove dependencies from GIC, and with this the correct behaviour for all families will be considered, including FPU lazy mode, but this is not easy and I currently have other priorities.

mlyle, 2016-08-30:
I think for now you should just not do the lazy mode-- just unconditionally preserve all registers.  It will have the correct outcome for almost all code (just using a few bytes more irq stack; unless it inspects its own stack in pretty deep ways)-- and can be enhanced to be more correct later.

ilg-ul, 2016-08-30:
I don't know, it might be so, but I need to analyse the issue thoroughly.

honzikm, 2017-06-01:
Any news about the M4 FPU support?

ilg-ul, 2017-06-01:
unless someone will decide to sponsor QEMU development, I'm afraid it'll take a while to get the FP support.

mlyle, 2017-06-01:
It would be really nice if there was a workaround-- to output a sternly worded warning and to save all the FP registers.  Almost all existing code would then work (main consequence would be a performance penalty).

honzikm, 2017-06-01:
I think I can offer some programming work.

ilg-ul, 2017-06-01:
> I can offer a bit of my time to do some programming

great!

did you take a look at the qemu source code?

lujnan, 2019-02-27:
I'm a newbie, What can I do about this subject?  I don't have any experience in either 'qemu' or 'mcu'.

ilg-ul, 2019-02-27:
well, first of all, continue learning. did you run the blinky tutorial till the end? did you see the graphical leds blinking?

then take a look at qemu source code.