ilg-archived/qemu #4: sld-image-1.2.10-setjmp patch missing when building qemu

dimdimdimdim, 2015-11-09:
Hi,
I wish to add some peripherals to the STM32 mcus, and I was trying to build QEMU from sources using the method described (using Docker).
The build-qemu.sh script fails with the following message:
`build-qemu.sh: line 608: /home/dim/Work/qemu/gnuarmeclipse-qemu.git/gnuarmeclipse/patches/sdl-image-1.2.10-setjmp.patch`
And the file is actually missing, there seems to be only two patches for the SDL-1.2.15 lib in that folder.

By the way, can this fork of qemu be used in standalone, or does it require the whole eclipse environment and the appropriate eclipse plugins to run ?

ilg-ul, 2015-11-09:
> the file is actually missing

are you sure you are using the gnuarmeclipse-dev branch?

https://github.com/gnuarmeclipse/qemu/tree/gnuarmeclipse-dev/gnuarmeclipse/patches

> can this fork of qemu be used in standalone?

yes

dimdimdimdim, 2015-11-10:
Hi,

Thank you for your answer

Well, it seems i'm actually using the right branch:
```
dim@Ubuntu64:~/Work/qemu/gnuarmeclipse-qemu.git$ git status
On branch gnuarmeclipse-dev
Your branch is up-to-date with 'origin/gnuarmeclipse-dev'.
```
Well, I don't really know how git works yet, anyway. So I got the patch file manually from the URL you gave and applied it myself (I first tried without it but compilation failed). Maybe my environment will be a bit wobbly, we'll see. Compilation hasn't finished yet.

Note that I got another error:
`build-qemu.sh: line 715: MACPORTS_FOLDER: unbound variable`
I just removed the line where the MACPORTS_FOLDER definition is, rerun, and it started to compile things. For information, I'm using Ubuntu 14.04 64 bits.

Thanks, anyway, and congratulations for your work. I'll let you know if I manage to implement a few missing peripherals...

ilg-ul, 2015-11-10:
> I'm using Ubuntu 14.04 64 bits to compile, but my target is win32.

as far as I know (did not test it) Docker should also run on Windows, so, for win32 targets, you should not need Ubuntu.

ilg-ul, 2015-11-10:
> I'll let you know if I manage to implement a few missing peripherals...

even if you do, at the end you'll probably acknowledge that part of this task is not for humans, it should be automated by generating the peripherals automatically from ARM SVD files. only the actions need to be added manually.

this is the reason I gave up implementing all those processors for the moment. time permitting, I'll add the engine to create the peripherals from a JSON file with the peripheral definitions, and another script to generate the JSON from the ARM SVD.

then it'll be relatively easy to add new processors.

dimdimdimdim, 2015-11-10:
I've seen that Docker could run on windows, but I did not dare to try. It would certainly need tweaking the build-qemu.sh script a lot, anyway, so I went the simpler way.

Regarding the SVD: these files only describe the register locations/bit fields/etc.., right (I'm quite new to these concepts) ? So, as you say, the actions still have to be added manually, which is certainly what constitutes the most important part of the code to add (the identification of the register that is being read/written being just a switch/case statement, with a few additional manipulations for the bit fields). So is it worth automating this ?

ilg-ul, 2015-11-10:
> Docker could run on windows, but I did not dare to try

I'm running it on OS X without problems. on Linux it runs natively, on non-linux (OS X, Windows) it runs under a VM Ubuntu virtualised with Virtual Box. 

> It would certainly need tweaking the build-qemu.sh script a lot

well, yes and no, most of the build script runs inside Docker, and this needs no changes. the top layer script indeed would not run on Windows directly, but I guess it might run under bash on MSYS2. however I did not try it.

> the identification of the register that is being read/written being just a switch/case statement, with a few additional manipulations for the bit fields). So is it worth automating this ?

it depends on the emulation accuracy you expect. if you want to be sure all registers are there, and behave exactly as the originals, the 'few additional manipulation for the bit fields' become very complex and doing them manually is tedious and error prone. try to implement the 'switch/case' for all access sizes allowed (8/16/32 and aligned and unaligned) and you'll see what I mean.

dimdimdimdim, 2015-11-10:
oooook, now I get it...
Well, actually, I didn't plan to have such an accurate emulation. It is overkill for my needs. I could even use another microcontroller than the one I'm targetting on hardware, in which case I would adapt the drivers from my embedded OS. What I need, however, is an easy way to "plug" the external peripherals I need (keyboard/screen/io/...) to the qemu-emulated microcontroller. I'm more focusing on the whole platform than on the microcontroller itself, and if it is not behaving like the real hardware, it does not really matter in my case. I just wanna see my application running even if it requires some hacks in the embedded OS (I didn't expect it to be exactly the same as on the real hardware anyway).
Anyway, I'm gonna fork your project and give it a try...
Thanks again

ilg-ul, 2016-02-17:
any progress with this? can we close this ticket?

dimdimdimdim, 2016-02-17:
Yes. I still don't know why it happened, but it could be fixed easily anyway.
There is my pull request still pending, however.

closed 2016-02-17 by ilg-ul