ilg-archived/qemu #7: host-guest socket-based communication support

ludico78, 2016-01-07:
Hello,
I'm interested in early-stage prototyping and unit testing of bare-metal Cortex-M projects and I'd like to move from an old-fashioned ARM simulator/emulator to qEMU (on Windows).
 
The former simulator allowed me to set-up socket-based host-guest communication and I used this feature to easily bridge host applications with the simulator without needing to have fully working low-level drivers in my prototype (example use case:host applications will rely on actual I2C drivers when connected to a real embedded product, but they also supports simple sockets to transfer their payloads when they are used to debug an early-stage product with no low-level driver for I2C). 

Now I'm looking for a simple way to realize the same socket-based host-guest communication on qEMU.
I know from original documentation that '-serial tcp:host:port [,server]' options allow intercommunication between the emulated target's serial ports and the host OS using TCP sockets. Is this feature also available on gnuarmecliplse/qemu? In that case how does it work?

Do you suggest me another way to realize the same behavior? Usage of semihosting could be an alternative but, from what I understood, it only works with files.

Thanks in advance and best regards,
Luca.

ilg-ul, 2016-01-07:
> -serial tcp:host:port [,server] ... Is this feature also available

unfortunately not.

> ... another way... semihosting

that's correct, semihosting works only with files.

there are some remote plans to extend qemu, but they have low priority.

ludico78, 2016-01-10:
Ok, many thanks for your answer!

In that case I would try the semihosting, even if, for me, host-guest communication using files is really the last option.

I saw that 'SYS_SYSTEM' semihosting syscall can be used for launching a process on the host. In that case I would still have a chance of recreating socket communication by invoking cmd-line network tools like 'netcat' etc..

Unfortunately I didn't see any usage example on 'SYS_SYSTEM'.
Is it supported on current version?

Rgds,
Luca.

ilg-ul, 2016-01-10:
I checked in qemu and the implementation is

```
    case TARGET_SYS_SYSTEM:
        GET_ARG(0);
        GET_ARG(1);
        if (use_gdb_syscalls()) {
            return arm_gdb_syscall(cpu, arm_semi_cb, "system,%s",
                                   arg0, (int)arg1+1);
        } else {
            s = lock_user_string(arg0);
            if (!s) {
                /* FIXME - should this error code be -TARGET_EFAULT ? */
                return (uint32_t)-1;
            }
            ret = set_swi_errno(ts, system(s));
            unlock_user(s, arg0, 0);
            return ret;
        }
```

we are using the `else` branch, where the `system()` call seems to be issued.

in my `syscalls.c` implementation, the semihosting part, the `SYS_SYSTEM` call also seems to be fully implemented, so... there are good chances for everything to work.

dimdimdimdim, 2016-01-11:
Hi,

Well, with the serial port implementation I provided in the pull request 
https://github.com/gnuarmeclipse/qemu/pull/6, the -serial option should 
work. I did not test it much, but it should work. Then, you just 
send/receive data through the corresponding USART peripheral.

But I don't know if these changes will make it to the official 
gnuarmeclipse qemu branch, though. This pull request has been there for 
quite some time but I did not have any feedback from Liviu. Liviu ? What 
do you think ? If you don't want to merge it, you should probably close 
it, anyway. In this cas, Luca, you can eventually build my qemu fork by 
following the build instructions available on the gnuarmeclipse web site.

Regards

Dimitri

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

ilg-ul, 2016-01-11:
@dimdimdimdim, I apologise, I completely missed your request.

I already planned a new QEMU release, possibly this week, so I'll take a look at your patch and possibly apply it.

ludico78, 2016-01-11:
@ilg-ul  
Starting from a 'blinky' project on STM32F4-Discovery (device: STM32F429ZI) I had a look at the `trace_puts` actually map to semihosting calls and so, i tried to do the same with the SYS_SYSTEM syscall.

I just included `arm/semihosting.h` and added the following line in the main() method:

`call_host (SEMIHOSTING_SYS_SYSTEM, (void*) "cmd.exe");`

I'm new to the project and maybe I'm doing the wrong things...but while everything works fine when the `trace_puts` are invoked, apparently nothing happens when such a line is being executed.

Any ideas?

ilg-ul, 2016-01-11:
instruct the wizard to create a new project using semihosting syscalls and then you have access to all semihosting functionality, including the `system()` call.

ludico78, 2016-01-11:
I would have expected that `trace_puts()` didn't work if something was wrongly set with semihosting.

Anyway, in the new project wizard I used `Use system calls = Semihosting (POSIX system calls via host)`.
In addition my current Debug Configuration already has the `Enable ARM semihosting` set.

In my code I used both `call_host (SEMIHOSTING_SYS_SYSTEM, (void*) "cmd.exe");` and the sdtlib `system("cmd.exe")`.

What went wrong?

dimdimdimdim, 2016-01-11:
@ilg-ul No problem. I was surprised not to have any feedbacks from you, but I didn't want to bother you by following up. You seem to be quite busy.

ilg-ul, 2016-01-11:
> `trace_puts()` ... wrongly set with semihosting.
the trace setting and the posix setting are completely unrelated. trace uses only an output stream, that can be redirected to one of the multiple devices. full semihosting means all posix calls (open, read, write, etc) are forwarded to the host, via the debugger interface.

> `system("cmd.exe")`

what did you expect to happen? you just started the `cmd.exe` program, which probably quit instantly for lack of I/O. `system()` does not provide access to the stdin/stdout/stderr of the started process.

ludico78, 2016-01-11:
>... `system()` does not provide access to the stdin/stdout/stderr of the started process..

So this mean that any cmd-line process trying to open default I/Os will not work? 

My first option was `system("echo TEST > output.txt")` and it didn't work either (no file created).
Then I tried to launch a process with graphic interface (e.g. PuTTY.exe) without success.

Any clue?

ilg-ul, 2016-01-11:
you can trace the application to be sure the semihosting call is really issued.

> no file created

does qemu display the tracing messages? if not, be sure you start qemu with semihosting enabled.

where did you expect the file to be created? I guess the current working directory is set to the Debug/Release folder, but I'm not sure.

ludico78, 2016-01-13:
Hi Liviu,
Sorry for this late response.

> you can trace the application to be sure the semihosting call is really issued
You mean by checking the assembly code? I did and I found the `bkpt` opcode is correctly invoked but nothing happens.

As I said in my previous post, the tracing messages are correctly displayed as I already enabled the semihosting.

> where did you expect the file to be created? ...
Indeed I had the same thought. Nothing is created in Debug/Release folder.
I also tried to force creation of the file using absolute path:
`system("echo TEST > C:\\output.txt")` 

nothing happened :(

Do you think there's something wrong with `sys_system` semihosting implementation?

ilg-ul, 2016-01-13:
> something wrong with sys_system semihosting implementation?

never tried this on windows, might be.

ludico78, 2016-01-14:
Hi Liviu,
Finally I solved!

> you can trace the application to be sure the semihosting call is really issued
>> You mean by checking the assembly code? I did and I found the bkpt opcode is correctly invoked 
but nothing happens.

I got confused and I found that only the `call_host()` was actually calling the `bkpt` instruction, while the standard `system(char* s)` API wasn't. In particular `system()` apparently did nothing at all. 
So I figured out that the "system()" API didn't map to the implementation contained in the _syscalls.c provided in the project template.

I created the mapping between `system(char *s)` -> `_system(char *s)`  and everything worked fine!

Now I'm ready to do some experiments with qEMU.

Thanks a lot for your support. 
Luca.

ilg-ul, 2016-01-14:
> I figured out that the system() API didn't map to the implementation contained in the _syscalls.c 

ah, probably newlib is configured without HAVE_SYSTEM, and in this case `system()` returns an error.

thank you for the info, I'll try to fix this in a future version of the template.

ilg-ul, 2016-02-17:
@ludico78, if no more problems, can you close this ticket now?

ludico78, 2016-02-17:
Ok.

closed 2016-02-17 by ludico78

vjpr, 2016-07-05:
I found a simple way to achieve this using unix pipes on OSX.

**main.cpp**

```
#include <string.h>

FILE* fp;
char line[100];
printf("Hello World!\n");
fp = fopen("/tmp/log.txt", "w");
if (fp == NULL) return 1;

for (;;) {

  FILE* in;
  //in = stdin;
  in = fopen("inpipe", "rw");
  if (in == NULL) continue;

  while (fgets(line, 100, in) != NULL) {
    if (strncmp(line, "test", 4) == 0) {
      printf("yo!\n");
    }
    fprintf(fp, "%s", line);
    if (line[0] == 'q') return 0;
    if (feof(in)) break;
  }
  fclose(fp);

}

printf("Finished!");
```

- Compile with `OS_USE_SEMIHOSTING`.

- Run in your project dir:

```
mkfifo inpipe
```

- Type echo "test\n" > inpipe

- You should see "yo" printed to your console.

jmfriedt, 2017-05-14:
not sure if this thread is still up to date ... I am attempting to use gnuarmeclipse-qemu (v2.8.0-644-ge45e0e1-dirty) under GNU/Linux (4.8.0-2-amd64) and would like to display the serial port output from the command line. I know my program for STM32F103 is working properly since on Andre Beckus' port of qemu I get when running
.../arm-softmmu/qemu-system-arm -M stm32-p103 -serial stdio -serial stdio  -kernel usart_cm3.bin 
I get
(process:20289): GLib-WARNING **: /Host/Work/qemu/glib-2.51.0/glib/gmem.c:483: custom memory allocation vtable not supported
VNC server running on `::1:5900'
LED Off
1
LED On
2
which is the expected behaviour (blinking LED with the numbers being output on USART2 as implemented on the Olimex STM32-P103 board).
If I do the same with gnuarmeclipse/qemu the program just hangs (gdb tells me I am stuck in 
usart_send_blocking(USART2, c); of libopencm3)
./qemu-system-gnuarmeclipse -M STM32-P103 -d unimp,guest_errors -nographic -serial null -serial telnet:localhost:1234,server -kernel usart_cm3.bin 
and replacing the blocking usart write with the non blocking version, I indeed get
./qemu-system-gnuarmeclipse -M STM32-P103 -d unimp,guest_errors -nographic -serial null -serial telnet:localhost:1234,server -kernel usart_cm3.bin 
QEMU 2.8.0 monitor - type 'help' for more information
(qemu) QEMU waiting for connection on: disconnected:telnet:localhost:1234,server
NVIC: SCR and CCR unimplemented
[led:red off]
[led:red on]
[led:red off]
...
after telnet localhost 1234.
Is there an "easy" way of display the USART output on console ?

Thanks.

ilg-ul, 2017-05-14:
> Is there an "easy" way of display the USART output on console ?

the USART implementation in GNU ARM Eclipse QEMU is incomplete, I created the framework for all peripherals and someone else started the USART implementation, but abandoned in the middle of the road. :-(

unfortunately I do not have resources to complete it now...

jmfriedt, 2017-05-15:
just so that I can grasp the complexity of the task, would it be possible to merge the functional implementation of the USART of Andre Beckus' qemu fork in which the USART is functional, or is Eclipse expecting some sort of additional interface that is not implemented yet (i.e. other than -serial stdio or -serial telnet:server) ?

ilg-ul, 2017-05-15:
> would it be possible to merge the functional implementation of the USART of Andre Beckus' qemu fork

yes, I guess so. 

the framework to implement all peripherals is there, the USART instance is there, just that some callbacks are not implemented properly.

but the expected functionality is not to forward output to a console, it is to properly forward both in and out to a tty device.

it isn't a very big deal once you know how to do it, but otherwise qemu is a very complex project and the only true documentation is the source code.