ilg-archived/qemu #9: LEDs on OLIMEX STM32-P107 board are switched

tgbloks, 2016-02-26:
i already created a branch with a fix but since im not allowed to push i added the patch here:
`diff --git a/hw/arm/stm32-olimex-boards.c b/hw/arm/stm32-olimex-boards.c
index 28dc9c6..2a11657 100644
--- a/hw/arm/stm32-olimex-boards.c
+++ b/hw/arm/stm32-olimex-boards.c
@@ -172,7 +172,7 @@ static GPIOLEDInfo stm32_p107_leds_info[] = {
         .name = "green-led",
         .active_low = false,
         .colour_message = "Green",
-        .x = 171,
+        .x = 227,
         .y = 354,
         .w = 6,
         .h = 8,
@@ -182,7 +182,7 @@ static GPIOLEDInfo stm32_p107_leds_info[] = {
         .name = "yellow-led",
         .active_low = false,
         .colour_message = "Yellow",
-        .x = 227,
+        .x = 171,
         .y = 354,
         .w = 6,
         .h = 8,
`

I also fixed an issue with malloc tracepoints and new glib versions:
(see https://lists.gnu.org/archive/html/qemu-devel/2015-09/msg01262.html)
`diff --git a/vl.c b/vl.c
index 9424767..56c8e85 100644
--- a/vl.c
+++ b/vl.c
@@ -2852,6 +2852,7 @@ static const QEMUOption *lookup_opt(int argc, char **argv,
     return popt;
 }

+#if !GLIB_CHECK_VERSION(2, 44, 0)
 static gpointer malloc_and_trace(gsize n_bytes)
 {
     void *ptr = malloc(n_bytes);
@@ -2872,6 +2873,26 @@ static void free_and_trace(gpointer mem)
     free(mem);
 }

+static void register_glib_mem_trace_functions(void)
+{
+    GMemVTable mem_trace = {
+        .malloc = malloc_and_trace,
+        .realloc = realloc_and_trace,
+        .free = free_and_trace,
+    };
+
+    g_mem_set_vtable(&mem_trace);
+}
+#else
+static void register_glib_mem_trace_functions(void)
+{
+    /* Newer versions of glib don't support intercepting the
+     * memory allocation functions.
+     */
+}
+#endif
+
+
 static int machine_set_property(void *opaque,
                                 const char *name, const char *value,
                                 Error **errp)
@@ -3099,11 +3120,6 @@ int main(int argc, char **argv, char **envp)
     bool userconfig = true;
     const char *log_mask = NULL;
     const char *log_file = NULL;
-    GMemVTable mem_trace = {
-        .malloc = malloc_and_trace,
-        .realloc = realloc_and_trace,
-        .free = free_and_trace,
-    };
     const char *trace_events = NULL;
     const char *trace_file = NULL;
     ram_addr_t maxram_size;
@@ -3146,7 +3162,7 @@ int main(int argc, char **argv, char **envp)
     }
 #endif /* defined(CONFIG_VERBOSE) */

-    g_mem_set_vtable(&mem_trace);
+    register_glib_mem_trace_functions();

     module_call_init(MODULE_INIT_QOM);

`

ilg-ul, 2016-02-26:
I added you as a contributor, please push the changes on a separate branch, each issue as a separate push.

ilg-ul, 2016-02-26:
btw, it seems you did not fork the project before trying to generate pull requests.

tgbloks, 2016-02-26:
thx for the hint. never used github before...

ilg-ul, 2016-04-04:
the pull request should be integrated for now, please confirm it is ok.

ilg-ul, 2016-07-27:
fixed in 2.6.0-20160728

closed 2016-07-27 by ilg-ul