ilg-archived/qemu #11: Fix of little mistake in manual [How to install the QEMU binaries?]

sledgeh, 2016-03-07:
In manual [How to install the QEMU binaries?](http://gnuarmeclipse.github.io/qemu/install/), section GNU/Linux 
> To check if QEMU starts and is recent, use:
```
$ /opt/gnuarmeclipse/qemu/debian64-2.2.92-201504041716-dev/bin/qemu-system-gnuarmeclipse.exe" --version
GNU ARM Eclipse 64-bit QEMU emulator version 2.2.92
Copyright (c) 2003-2008 Fabrice Bellard
```
First line of code should be:
```
$ /opt/gnuarmeclipse/qemu/2.4.50-201510290935-dev/bin/qemu-system-gnuarmeclipse --version
```
FIXED.

ilg-ul, 2016-03-07:
thank you, I fixed the manual page, it'll be available soon.

if you discover more problems with the documentation, please use the dedicated tracker : https://github.com/gnuarmeclipse/gnuarmeclipse.github.io/issues/

sledgeh, 2016-03-07:
ok. Thank you for your work.

closed 2016-03-08 by ilg-ul