ilg-archived/qemu #13: STM32F411 eval board LED problems

mlyle, 2016-04-23:
It doesn't seem possible to turn off the LED with 
```
GPIO_WriteBit(GPIOA, GPIO_Pin_5, toggle ? Bit_SET : Bit_RESET);
```

from the stdperiph library, which works on hardware.   Using SET to turn it on works.  `GPIO_ToggleBits(GPIOA, GPIO_Pin_5);` works to turn it on and off.  Is there something wrong with the implementation of the BSRRL register?

My command line (and version) are:

```
/Applications/GNU\ ARM\ Eclipse/QEMU/2.4.50-201510290935-dev/bin/qemu-system-gnuarmeclipse -board NUCLEO-F411RE -image build/openloader.bin
```

It looks like reset bits just does a half-word store:

```
080105ba <GPIO_ResetBits>:
 80105ba:       8341            strh    r1, [r0, #26]
 80105bc:       4770            bx      lr
```

Is the problem that #26 != #24  (doesn't line up with BSRR)?

ilg-ul, 2016-04-23:
I'll investigate, but it'll take some time.

ilg-ul, 2016-07-25:
I'm afraid I don't understand your report.

I just ran a blinky test on the emulated NUCLEO-F411RE board and the green LED turns on and off as expected.

Setting bits in the lower half of BSSR turns the LED on, setting bits in the upper half of BSSR turns the LED off.

Isn't this the expected behaviour?

mlyle, 2016-07-25:
This is the exact code that fails on the emulator and works on hardware:

https://github.com/mlyle/openlager/blob/mpl-fail/loader/openloader.c#L113

Attached is the raw code.

[binfile.zip](https://github.com/gnuarmeclipse/qemu/files/382108/binfile.zip)

I think it fails because there's a half-word store (strh) to \#26, rather than a full-word store to the register address \#24

ilg-ul, 2016-07-25:
> a half-word store

aha! got it now. I'll check.

ilg-ul, 2016-07-26:
I identified the problem, it was a misunderstanding in passing the value in the peripheral register implementation. I fixed this by adding a new parameter `full_value` which is the entire register, and used it in the BRR register emulation.

https://github.com/gnuarmeclipse/qemu/commit/add4dc4807f2c9804944c1abf3c53e2c79bbdb2f

do you build qemu locally, to test the latest version, or prefer to wait for the next official release?

ilg-ul, 2016-07-27:
fixed in 2.6.0-20160728

closed 2016-07-27 by ilg-ul