ilg-archived/qemu #14: Trouble with QEMU

jzsig, 2016-04-29:
When I go through the blinky tutorial and try to debug with QEMU, I get the following error:

Could not determine GDB version after sending: arm-none-eabi-gdb --version, response: 

I made sure that the toolchain was pointed to the correct location, and it is, so I'm not sure what the problem is.

jzsig, 2016-04-29:
I fixed the above problem by installing the correct 32 bit drives for gdb, but now I'm getting an error:

An internal error occured during: "Launching blinky".
org.eclipse.cdt.dsf.gdb.service.GdbDebugServicesFactory.<init>(Ljava/lang/String;)V

ilg-ul, 2016-04-29:
In the README FIRST issue #1 are listed some details regarding your configuration.

I do not know what '32-bit drives for gdb' you are talking about.

jzsig, 2016-04-29:
Sorry about that, here is the requested info:

QEMU Version: 2.4.50
System Version: Centos 2.10 64-bit

Here is screenshot:
[http://imgur.com/y6nvyr1](url)

I have actually determined that it is not QEMU its self since I can use the command line interface and it works i.e. using 

./qemu-system-gnuarmeclipse -board STM32F4-Discovery -mcu STM32F407VG --image /home/jzsig/workspace/blinky/Debug/blinky.elf 

However, when I try and debug from eclipse, it does nothing and gives the GDB error above. The 32-bit drivers I was referring to 32 bit libraries that were needed to run GDB.

ilg-ul, 2016-04-29:
> Command Line: Sent from eclipse, not sure how to find exact command.

right click on the qemu process in the Debug tab, the Debug perspective

> Here is screenshot:

this looks like an Eclipse problem.

check the Eclipse log (workspace/.metadata/.log) for more info; check the definitions of the qemu path and executable macros, maybe you have a wrong character there.

jzsig, 2016-04-29:
Yeah it was a problem with eclipse. I resolved the problem by switching to Mars instead of Neon.

ilg-ul, 2016-04-30:
so the conclusion is that the qemu plug-in does not run on Neon? can you try again with a fresh Neon, without any other plug-ins or custom settings?

jzsig, 2016-05-01:
Correct, thats what it seems like. I did a freash install received the same problem. It was giving the same error above for the qemu plugin as well as the j-link plugin.

ilg-ul, 2016-05-01:
thank you, in this case I have to investigate the plug-ins behaviour on Neon.

closed 2016-05-02 by jzsig

ilg-ul, 2016-07-25:
The Neon debugging issues were fixed in the previous release, can you confirm everything is ok now?