ilg-archived/qemu #16: QEMU failed with code (1)

xdpsee, 2016-06-17:
debug,  it show 

QEMU failed with code (1).
For more details, see the qemu-system-gnuarmeclipse console.

but i run from command line, it's okay

/Applications/GNU\ ARM\ Eclipse/QEMU/2.4.50-201510290935-dev/bin/qemu-system-gnuarmeclipse --board STM32-P103 --mcu STM32F103RB --verbose --image /Users/zhenhui/Documents/workspace/stm32p103/Debug/stm32p103.elf

GNU ARM Eclipse 64-bits QEMU v2.4.50 (qemu-system-gnuarmeclipse).
Board: 'STM32-P103' (Olimex Prototype Board for STM32F103RBT6).
Device: 'STM32F103RB' (Cortex-M3 r0p1, MPU), Flash: 128 kB, RAM: 20 kB.
Image: '/Users/zhenhui/Documents/workspace/stm32p103/Debug/stm32p103.elf'.
Command line: (none).
Cortex-M3 r0p1 core initialised.
Cortex-M3 r0p1 core reset.
main(argc=1, argv=[""]);
Hello ARM World!
Standard output message.
Standard error message.
System clock: 72000000 Hz
[Red LED Off]
[Red LED On]
[Red LED Off]
Second 1
[Red LED On]
[Red LED Off]
Second 2
[Red LED On]
[Red LED Off]
Second 3
[Red LED On]
[Red LED Off]
Second 4
[Red LED On]
[Red LED Off]
Second 5
QEMU exit(0)

what can i do?

ilg-ul, 2016-06-17:
the qemu console does not show anything?

the Eclipse logs?

to me it looks like one of the GUI fields is not correct and Eclipse cannot start the executable.

xdpsee, 2016-06-21:
i shouldn't use the eclipse 4.5, it's ok when use eclipse 4.4

closed 2016-06-21 by xdpsee

ilg-ul, 2016-06-21:
I use qemu with eclipse 4.5 daily. it might be something else.