ilg-archived/qemu #17: qemu-system-gnuarmeclipse process not terminating on exit

nscooling, 2016-06-29:
I am running a qemu session from the command line as per:
http://gnuarmeclipse.github.io/qemu/options/

My command line is:
$ qemu-system-gnuarmeclipse -verbose -verbose --mcu STM32F407VG -d unimp,guest_errors --nographic --image ./qemu_blinky/Debug/qemu_blinky.elf --semihosting-config enable=on,target=native 

The application runs fine:
GNU ARM Eclipse 64-bits QEMU v2.4.50 (qemu-system-gnuarmeclipse).
Board: 'generic' (Generic Cortex-M board; use -mcu to define the device).
Device: 'STM32F407VG' (Cortex-M4 r0p0, MPU), Flash: 1024 kB, RAM: 128 kB.
Image: './qemu_blinky/Debug/qemu_blinky.elf'.
Command line: (none).
Load   8613 bytes at 0x08000000-0x080021A4.
Load    120 bytes at 0x080021A8-0x0800221F.
Load    420 bytes at 0x20000078-0x2000021B.
Cortex-M4 r0p0 core initialised.
QEMU 2.4.50 monitor - type 'help' for more information
(qemu) QEMU 2.4.50 monitor - type 'help' for more information
(qemu) Cortex-M4 r0p0 core reset.
PRIGROUP unimplemented
Hello ARM World!
System clock: 16000000 Hz
Second 1
Second 2

etc. 
I have set the loop to count 20 times and exit main with return 0. 
However on exiting the process does not return to the CLI and (I believe) sits in the exit while(1) loop.

Is there a pre-processor setting missing/required?

This is running on OSX.

ilg-ul, 2016-06-29:
your application must be fully semihosted.

nscooling, 2016-06-30:
Worth mentioning that on the page:
http://gnuarmeclipse.github.io/qemu/options/

ilg-ul, 2016-07-01:
done. please check http://gnuarmeclipse.github.io/qemu/options/#fully-semihosted-applications

closed 2016-07-27 by ilg-ul