ilg-archived/qemu #19: STM32 registers reset

npereiras, 2016-07-18:
I'm trying to enable the usart1 for use with the host device.
During my test I found out that the SR register value is not been set to the reset value.

I tried to correct the problem but I'm not sure if this is the way of doing it. 

Can you check my patch below?

[0001-Correct-register-reset-process-and-add-uart-reset.txt](https://github.com/gnuarmeclipse/qemu/files/369780/0001-Correct-register-reset-process-and-add-uart-reset.txt)

ilg-ul, 2016-07-18:
it seems ok, it builds on my machine.

the first part of the patch, fixing `peripheral_register_reset_callback()` looks like a bug affecting all registers, it is hard to believe how it worked before...

does it work now in your environment?

npereiras, 2016-07-19:
Yes, on my environment it seems to work correctly, but I only did simple tests in Non Blocking mode APIs with Interrupt.

ilg-ul, 2016-07-19:
ok, I pushed the changes.

ilg-ul, 2016-07-27:
fixed in 2.6.0-20160728

closed 2016-07-27 by ilg-ul