ilg-archived/qemu #22: How to open a file with O_NONBLOCK flag set

vjpr, 2016-07-24:
My issue is that when using `fgets` to read from a unix fifo on the host, the semihosted `read` syscall is blocking all execution (including my FreeRTOS threads).

So it seems like I need to open a file with the flag `O_NONBLOCK`.

However the host-side `TARGET_SYS_OPEN` seems to prevent passing anything but basic flags:

```
ret = set_swi_errno(ts, open(s, open_modeflags[arg1], 0644));
```

https://github.com/gnuarmeclipse/qemu/blob/d9d9f8e848c364b0a923bb178c2361ea1e6ac40f/target-arm/arm-semi.c#L311

What is the reason for this?

---

Also, when using `fopen`, `fcntl` is the recommended approach for setting this flag. How difficult would it be to support the `fcntl` syscall? Could it just be added as a command to the qemu `arm-semi.c` and newlib `_syscalls.c`?

```
int fd = fileno(in);
int flags;
flags = fcntl(fd, F_GETFL, 0);
flags |= O_NONBLOCK;
fcntl(fd, F_SETFL, flags);
```

ilg-ul, 2016-07-24:
did you check the ARM semihosting specs? are these flags supported?

if so, can you provide a patch with your suggestions?

vjpr, 2016-07-24:
Regarding, `fcntl`, looks like its a very limited api: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0471g/Bgbjjgij.html

Is there a reason you implemented `open_modeflags[arg1]` instead of passing through the flags? Are there limits on the argument sizes in the semihosting spec?

ilg-ul, 2016-07-24:
> any reason we can't go outside the spec?

compatibility?

> limits on the argument sizes in the semihosting spec?

you mean command line arguments? if I remember right, there is a minimum limit.

ilg-ul, 2016-07-24:
>  the semihosted `read` syscall is blocking all execution (including my FreeRTOS threads)?

btw, by design the semihosting interface is not real time, since the debugger keeps the target halted during the calls. an RTOS (be it free or not) can do nothing to take control. as such, the semihosting interface is not particularly RTOS friendly.

vjpr, 2016-07-24:
Thanks, I am resorting to polling the host using RPC calls using `write` and then an immediate `read`, instead of trying to `read` from a unix fifo in a loop now.

ilg-ul, 2016-07-25:
FYI, if you are using FreeRTOS with QEMU, I just added BASEPRI support (#21).

vjpr, 2016-07-25:
> FYI, if you are using FreeRTOS with QEMU, I just added BASEPRI support (#21).

Awesome!

closed 2016-10-29 by ilg-ul