ilg-archived/qemu #24: --semihosting-cmdline with double-dash option doesn't work

vjpr, 2016-08-16:
*Command*

```
qemu-system-gnuarmeclipse  \
--gdb tcp::1234 \
-D qemu-log.txt \
-d unimp,guest_errors \
-monitor null \
-serial null \
--verbose \
--verbose \
--board NUCLEO-F411RE \
--mcu STM32F411RE \
--nographic \
--image bin/debug/app.elf \
--semihosting-config enable=on,target=native \
--semihosting-cmdline "--devMode=tag"
```

*Output*

```
GNU ARM Eclipse 64-bits QEMU v2.4.50 (qemu-system-gnuarmeclipse).
qemu-system-gnuarmeclipse: --devMode=tag: invalid option
```

Works:

`--semihosting-cmdline -d tag`

Variation which don't work:

`--semihosting-cmdline --devMode=tag` (no quotes)

It appears that when an `=` is used it doesn't work.

vjpr, 2016-08-16:
So my problem was that I had to pass the name of the app as the first argument without dashes.

E.g. `--semihosting-cmdline app -d tag`

This was already mentioned in the docs, but was only a recommendation. It seems that it *always* needs to be the app name.

vjpr, 2016-08-16:
Also, in the Eclipse UI I have to make the cli command input:

```
"app --devMode=tag "
```

Note the trailing space after tag and before the quote. Without this, the argument is getting parsed as `tag"`.

Also, if I remove the quotes from this input, QEMU attempts to parse them instead.

If I put a quote at the start, then a quote is automatically added to the end which gets included in `argv`.

ilg-ul, 2016-08-16:
thank you for the report, I'll investigate.

ilg-ul, 2016-12-27:
fixed in 2.8

closed 2016-12-27 by ilg-ul