ilg-archived/qemu #25: GNU ARM Eclipse QEMU v2.6.0-20160728 on Ubuntu 14.04LTS

ivlz, 2016-09-04:
I followed the instructions at http://gnuarmeclipse.github.io/qemu/install/ and installed v2.6.0-20160728. However it fails to launch in Eclipse and on the command line. 

Eclipse GUI reports:
"QEMU failed with code (-1).
For more details, see the qemu-system-gnuarmeclipse console."

Eclipse QEMU console reports:
GNU ARM Eclipse 64-bits QEMU v2.6.0 (qemu-system-gnuarmeclipse).
Board: 'generic' (Generic Cortex-M board; use -mcu to define the device).
Device: 'STM32F103RB' (Cortex-M3 r0p1, MPU), Flash: 128 kB, RAM: 20 kB.
Command line: 'testqemu' (8 bytes).
Cortex-M3 r0p1 core initialised.

Commandline reports:
>/opt/gnuarmeclipse/qemu/2.6.0-201607280535/bin/qemu-system-gnuarmeclipse -board generic -mcu STM32F103RB --image testqemu.elf
qemu-system-gnuarmeclipse: /Host/Work/qemu/gnuarmeclipse-qemu.git/hw/cortexm/peripheral.c:281: peripheral_realize_callback: Assertion `state->max_offset_bytes < state->mmio_size_bytes' failed.
Aborted (core dumped)

I then followed the same instructions to install v2.4.50-201510290935-dev. This works perfectly in both Eclipse and the command line.

Is it possible there is a bug in the latest release or have I misstepped somewhere?

Thanks.

MACHINE SETUP:
>uname -a
Linux workstation0 4.4.0-36-generic #55~14.04.1-Ubuntu SMP Fri Aug 12 11:49:30 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

>ls /opt/gnuarmeclipse/qemu
2.4.50-201510290935-dev  2.6.0-201607280535

ilg-ul, 2016-09-04:
I confirm the issue, it is a bug.

in the latest release I applied a patch to initialise all available USARTs in the device, and this worked properly for F4, but for F1 the USARTs are not yet implemented, and the assert triggered.

the workaround was to temporarily remove the capabilities related to USARTs in all other devices than F4.

the repository is up-to-date now, it you want you can build the Linux version.

otherwise I plan to update QEMU to 2.7 in a few weeks time and make a new release, which will include this patch too.

ivlz, 2016-09-05:
Thank you for the prompt and helpful response.

ilg-ul, 2016-11-28:
fixed in v2.7.0-20161029

closed 2016-11-28 by ilg-ul

ivlz, 2016-12-04:
Hi Liviu
    Thanks for hthe fix. It works.
    
    Regards
    Inno.
    
    On 28/11/16 21:27, Liviu Ionescu wrote:
    
    
      Closed #25.
      —
          it on GitHub, or mute
          the thread.

ivlz, 2016-12-04:
Thanks for the fix. I can confirm that it works.

ilg-ul, 2016-12-04:
FYI, some major internal changes are under way, so I would appreciate if you could also test the future releases.

ilg-ul, 2016-12-29:
could you check the latest release (2.8) and confirm if it is ok?