ilg-archived/qemu #26: gnuarmeclipse-softmmu build error

yifanz, 2016-09-27:
I'm trying to do a `bash build-qemu.sh --debian64` as per the build instructions. I'm getting the following build error. Appears that boot.o may need to link with device_tree.o, but device_tree.o isn't even compiled. I didn't change build configurations and I'm on the `bash build-qemu.sh checkout-stable` branch.

hw/arm/boot.o: In function `load_dtb':
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:417: undefined reference to `load_device_tree'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:441: undefined reference to `qemu_fdt_getprop_cell'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:443: undefined reference to `qemu_fdt_getprop_cell'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:459: undefined reference to `qemu_fdt_setprop_sized_cells_from_array'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:496: undefined reference to `qemu_fdt_dumpdtb'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:468: undefined reference to `qemu_fdt_setprop_string'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:477: undefined reference to `qemu_fdt_setprop_cell'
/Host/Work/qemu/gnuarmeclipse-qemu.git/hw/arm/boot.c:484: undefined reference to `qemu_fdt_setprop_cell'
collect2: error: ld returned 1 exit status
make[1]: *** [qemu-system-gnuarmeclipse] Error 1
Makefile:195: recipe for target 'qemu-system-gnuarmeclipse' failed
make: *** [subdir-gnuarmeclipse-softmmu] Error 2
**Makefile:186: recipe for target 'subdir-gnuarmeclipse-softmmu' failed**

ilg-ul, 2016-09-27:
I did not build the stable branch for some time, could you try the current development branch?

yifanz, 2016-09-27:
Development branch build successfully. Thanks!

bash build-qemu.sh cleanall
**bash build-qemu.sh checkout-dev**
bash build-qemu.sh --debian64

closed 2016-09-27 by yifanz

ilg-ul, 2016-09-27:
great!

I plan to update to 2.7 in the next few weeks, and then update the stable branch.