ilg-archived/qemu #27: gnuarmeclipse-qemu-osx-2.6.0-201607280535.pkg missing liblzo shared library

bjandre, 2016-10-06:
The macOS installer package for gnuarmeclipse-qemu-osx-2.6.0-201607280535 is missing the liblzo shared library:

```
$ /Applications/GNU\ ARM\ Eclipse/QEMU/2.6.0-201607280535/bin/qemu-system-gnuarmeclipse --version
dyld: Library not loaded: /opt/macports/lib/liblzo2.2.dylib
  Referenced from: /Applications/GNU ARM Eclipse/QEMU/2.6.0-201607280535/bin/qemu-system-gnuarmeclipse
  Reason: image not found
Trace/BPT trap: 5
```
liblzo2.2.dylib should probably be included in the libraries included in:
`/Applications/GNU\ ARM\ Eclipse/QEMU/2.6.0-201607280535/bin/`.

A work around is to install liblzo from another source. For example:
```
$ brew install lzo
...
...
$ /Applications/GNU\ ARM\ Eclipse/QEMU/2.6.0-201607280535/bin/qemu-system-gnuarmeclipse --version
GNU ARM Eclipse 64-bits QEMU emulator version 2.6.0
Copyright (c) 2003-2008 Fabrice Bellard
```

ilg-ul, 2016-10-06:
thank you for the bug report, I'll fix it with the new release.

what os version are you using?

bjandre, 2016-10-06:
I'm running:
* OS X El Capitan, v10.11.6.
* xcode 8.0 (8A218a)
* homebrew instead of macports.

ilg-ul, 2016-10-17:
I just fixed the build scripts (https://github.com/gnuarmeclipse/build-scripts), could you try to build your own qemu and let me know if the scripts and the resulting package are ok?

alternatively I have to send you a .pkg file, but I wouldn't make a new release only for this, I plan to upgrade to 2.7 soon.

ilg-ul, 2016-10-17:
actually I added the binary files to the 2.6.0 release, check the ones ending in `-dev`.

adamvn, 2016-10-17:
Hi thanks,

Busy trying to get my mac setup for build.. that seems much simpler.

-Adam

Disclaimer - University of Cape Town This e-mail is subject to UCT policies and e-mail disclaimer published on our website at http://www.uct.ac.za/about/policies/emaildisclaimer/ or obtainable from +27 21 650 9111. If this e-mail is not related to the business of UCT, it is sent by the sender in an individual capacity. Please report security incidents or abuse via csirt@uct.ac.za

adamvn, 2016-10-17:
great, -dev works just fine.

ilg-ul, 2016-10-17:
please do not reply by mail to these messages, since this requires manual editing of the posts to remove the quoted parts.

xprite, 2016-10-24:
This is my experience trying to run qemu on El Capitan 10.11.6 (15G31)

$ ./qemu-system-gnuarmeclipse --version
dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
  Referenced from: /Applications/GNU ARM Eclipse/QEMU/2.6.0-201610170917-dev/bin/./qemu-system-gnuarmeclipse (which was built for Mac OS X 10.12)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _clock_gettime
  Referenced from: /Applications/GNU ARM Eclipse/QEMU/2.6.0-201610170917-dev/bin/./qemu-system-gnuarmeclipse (which was built for Mac OS X 10.12)
  Expected in: /usr/lib/libSystem.B.dylib

Trace/BPT trap: 5

ilg-ul, 2016-10-24:
hmmm... it looks like applications built for 10.12 are no longer running on 10.11.

unfortunately all my Macs are now at 10.12; do you know any trick to generate applications for 10.11?

xprite, 2016-10-24:
I have no idea. I just updated to macOS Sierra and QEMU is now fully functional.

ilg-ul, 2016-10-24:
> it looks like applications built for 10.12 are no longer running on 10.11.

unfortunately this is a know problem of Xcode 8 :-(

https://bugs.erlang.org/browse/ERL-256

ilg-ul, 2016-10-25:
the current workaround seems to be to completely avoid `clock_gettime()` on macOS builds.

if you still have a 10.11 running somewhere, I can provide a binary.

xprite, 2016-10-25:
I am good with 10.12 now. Maybe somebody else will be interested. Thank you

ilg-ul, 2016-11-28:
fixed in v2.7.0-20161029

closed 2016-11-28 by ilg-ul