ilg-archived/qemu #28: Keep qemu alive after program finishes

vjpr, 2016-10-09:
Is there a way to keep qemu alive after the program finishes?

I am writing a test watcher, and would like to use gdb to quickly reflash and system_reset, instead of restarting qemu.

vjpr, 2016-10-09:
Nevermind, I forgot about `_startup.c` which calls `exit`.

closed 2016-10-09 by vjpr

vjpr, 2016-10-09:
Still, it would be nice to control this behaviour with recompiling from outside the firmware...is there a way?

ilg-ul, 2016-10-10:
I'm not sure I understand your problem, but if you want QEMU and the debugger to remain active when the program terminates, you need to issue an BKPT in exit.

for example the bare metal version of the exit function is implemented with `NVIC_SystemReset()`

https://github.com/micro-os-plus/micro-os-plus-iii/blob/xpack/src/libc/stdlib/exit.c

while the semihosting version issues a specific BKPT (around line 1069):

https://github.com/micro-os-plus/micro-os-plus-iii/blob/xpack/src/semihosting/c-syscalls-semihosting.cpp

and the exception handlers end with a BKPT(0):

https://github.com/micro-os-plus/micro-os-plus-iii/blob/xpack/src/startup/exception-handlers.c

vjpr, 2016-10-10:
> you need to issue an BKPT in exit.

Thanks, this is a good idea and resolves my problem.

It looks like this will break only if the debugger is connected, which is basically what I am looking for.

I can also use semihosting cli args to determine whether the bkpt instruction is used.

ilg-ul, 2016-10-10:
> I can also use semihosting cli args

yes. you can take a closer look at the µOS++ sources, all these details were considered there.

you can also take a look at the µOS++ demo projects (https://github.com/micro-os-plus/eclipse-demo-projects). the generate.sh script is certainly functional on macOS, for linux I got a report that the latest version fails (I'll to fix it, hopefully today).

ilg-ul, 2016-10-10:
the `generate.sh` scripts are now functional on macOS, GNU/Linux and Windows.