ilg-archived/qemu #29: Connection between two QEMU MCUs

T0jan, 2016-10-13:
Hey there, 
This more an question than an real issue. For testing purposes I try to connect two STM32 emulated with the GNU ARM Eclipse QEMU over UART with each other. The code is already tested with two OLIMEX STM32-P103 boards. The reason i want to realize this also with QEMU is that i want to test the UART communication between this two MCUs already if the boards I want to put them on aren't finished.   
So now to my question: I'm a newbie in QEMU but if I understand it right the `-serial` addition to the command line connects one of the serial ports of the MCU with the port defined afterwards. My idea was to use `-serial pty` on the master MCU to etablish a new virtual port for this conversation and then connect the slave MCU via `-serial /dev/pts/XX` to that port. On the good side the starting process with both MCUs on QEMU runs without errors. But there happens no communication between these two both hanging in their request sending state but doesn't trigger any other action. So is it possible to handle my tries on this way?  
In relation to this I also have a question to the GDB session the plug-in starts: In which way it establish its connection to the MCU and does it maybe use the USART1 port?
And in addition I want to ask also if it is possible to get a view on the full output of QEMU. Because if I want to execute standard QEMU commands like `info ` or even `help` QEMU never generates any output.
Thanks.

ilg-ul, 2016-10-13:
your idea to connect two emulated devices via pty devices is interesting, but I do not know how far went the implementation to support this.

unfortunately I do not have any documentation on this, you need to check the qemu source code. limited usart support for some stm32 devices was contributed by users, and I do not have experience using it.

if I were in your place, I would do the following:

- write two posix programs, running on mac or linux, to communicate over a pty.
- rewrite one of these two programs to run on cortexm, and try to communicate with the other native project
- if problems arise, fix the qemu usart implementation
- when the first half is functional, rewrite the second program to run on cortexm, and try to communicate with the other qemu
- if problems arise, fix the qemu usart implementation

T0jan, 2016-10-13:
thanks for the fast response, i will try to get some results on the way you recommended and I will share the fixes for qemu here.

Or do you may have have other ideas to realize an serial connection between two STM32 virtual?

ilg-ul, 2016-10-13:
yes, please share your experience with other users.

for general discussions please use the project forum, and keep this tracker for bugs and enhancements only.

T0jan, 2016-10-18:
ok I could set up an basic setup in the way I told on top with creating an new PTY and then connect with the second device to it. I used an light modified version of beckus STM32 QEMU which seems to be also the foundation of your STM32 implemention. 

So with this dedicated version it works fine, but if I want to reproduce the results with your plug-in, UART doesn't work. I think it has something to do with the UART handling of the plug-in which caused you some trouble before.

I tried it with the 2.4.50-201510290935-dev build of QEMU which doesn't had the mentioned problems with UART.

ilg-ul, 2016-10-18:
the UART code in GNU ARM Eclipse QEMU was contributed by someone else and I had no time to check it.

so, if you identify problems and can find solutions to fix it, please let me know.

during the next days I plan to update to qemu 2.7, and then perhaps I have a few days to fix some pending issues, before making a new release.

T0jan, 2016-10-18:
well, what I can say after a short look in the `stm32_usart.c` here in this QEMU version is, that it seems to have only functions for STM32F4xx and some status queries but nothing for a transmit with basic STM32 controllers. 
If I take a look in beckus UART implementation here: https://github.com/beckus/qemu_stm32/blob/stm32/hw/char/stm32_uart.c
I see way more functions for different purposes. I can't tell if the contributer of the UART code placed them somewhere else, but with my minimal knowledge about QEMU source code, I cant clear this point completely.

ilg-ul, 2016-10-18:
I currently do not have a serial port on my mac, so I'm afraid I cannot test any serial code, but if you want to contribute, I can review it.

T0jan, 2016-10-18:
it would be a pleasure for me to help you, but the only kind of contribution i can offer is testing any new version you offer for the serial implementation.

ilg-ul, 2016-10-18:
> it would be a pleasure for me to help

thank you, I appreciate it.

> new version you offer for the serial implementation.

given the lack of resources, I cannot commit on this, but perhaps you could contact the original contributor, maybe he can further improve the USART code.

closed 2016-10-29 by ilg-ul