ilg-archived/qemu #31: GPIO toggling does not work

ivlz, 2016-12-04:
I tried to run ST the GPIO_IOToggle example project from ST's library STM32Cube_FW_F1_V1.4.0. Everything was okay except that the function  HAL_GPIO_TogglePin(LED2_GPIO_PORT, LED2_PIN) in main.c at line 100 does not toggle the GPIO output. I ended up using HAL_GPIO_WritePin(LED2_GPIO_PORT,LED2_PIN, seconds % 2) and this fixed it.  I believe this may be a QEMU issue because even direct register access confirmed that the GPIO was not being toggled.

Steps to reproduce (Ubuntu 14.04LTS, Eclipse CDT Neon)
1. Load the project STM32Cube_FW_F1_V1.4.0/Projects/STM32F103RB-Nucleo/Examples/GPIO/GPIO_IOToggle into Eclipse, also pulliing in all the required sources/headers from 
2. Create a QEMU run configuration.
3. Run the project as is, and it will be observed no toggling takes place.
  Hello ARM World!
  Second 1
  [Green LED On]
  Second 2
  Second 3
  Second 4
  Second 5
  Second 6

4. Replace the line 100 of STM32Cube_FW_F1_V1.4.0/Projects/STM32F103RB-Nucleo/Examples/GPIO/GPIO_IOToggle/Src/main.c with HAL_GPIO_WritePin(LED2_GPIO_PORT,LED2_PIN, seconds % 2). Toggling should be observed to take place as expected:
  Hello ARM World!
  Second 1
  [Green LED On]
  Second 2
  [Green LED Off]
  Second 3
  [Green LED On]
  Second 4
  [Green LED Off]
  Second 5
  [Green LED On]
  Second 6
  [Green LED Off]

ilg-ul, 2016-12-04:
thank you for the bug report, I'll investigate.

dgtlrift, 2016-12-06:
I'm trying to follow the HelloWorld tutorial and QEMU crashes with it.  I get a few dialog boxes:

```
[QEMU machine emulators and tools]
QEMU machine emulators and tools has stopped working
A problem caused the program to stop working correctly. 
Windows will close the program and notify you if a solution is available.
[Close program]
```
Then the other dialog says:
```
[Problem Occurred]
'Launching hello Debug' has encountered a problem.
Error in services launch sequence
[OK][Details>>]
```
From the Console, the logs show:
```

GNU ARM Eclipse 64-bits QEMU v2.7.0 (qemu-system-gnuarmeclipse.exe).
Board: 'STM32-H103' (Olimex Header Board for STM32F103RBT6).
Device: 'STM32F407VG' (Cortex-M4 r0p0, MPU), Flash: 1024 kB, RAM: 128 kB.
Command line: 'hello 3 2 1' (11 bytes).
Cortex-M4 r0p0 core initialised.
Board picture: 'C:\Program Files\GNU ARM Eclipse\QEMU\2.7.0-201611282115-dev\images\STM32-H103.jpg'.
Unexpected error in object_property_find() at /Host/Work/qemu/gnuarmeclipse-qemu.git/qom/object.c:1007:
qemu-system-gnuarmeclipse.exe: Property '.unnamed-gpio-out[12]' not found

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
```

I just downloaded this off the instructions specified with [http://gnuarmeclipse.github.io/debug/qemu/](http://gnuarmeclipse.github.io/debug/qemu/) [http://gnuarmeclipse.github.io/plugins/packs-manager/](http://gnuarmeclipse.github.io/plugins/packs-manager/) - is there an update to the documentation that is needed to overcome this issue?

ilg-ul, 2016-12-06:
I suggest for the moment you use the STM32F4DISCOVERY board, which was recently updated to implement even the push buttons.

The H103 will be updated in a future release (working on QEMU right now).

dgtlrift, 2016-12-06:
Thanks - using STM32F4-Discovery solved my issue...

closed 2016-12-27 by ilg-ul

ilg-ul, 2016-12-29:
could you check the latest release (2.8) and let me know if it works?