ilg-archived/qemu #32: .unnamed-gpio-out[13]' not found

williamjamir, 2016-12-09:
Executing an elf file the following error is displayed.
 
Command:
qemu-system-gnuarmeclipse --verbose --board STM32F429I-Discovery --mcu STM32F429ZI -d unimp,guest_errors --nographic --image blinky.elf --semihosting-config enable=on,target=native --semihosting-cmdline test 1 2 3
 
Output:
GNU ARM Eclipse 64-bits QEMU v2.7.0 (qemu-system-gnuarmeclipse).
Board: 'STM32F429I-Discovery' (ST Discovery kit for STM32F429/439 lines).
Device: 'STM32F429ZI' (Cortex-M4 r0p0, MPU), Flash: 2048 kB, RAM: 192 kB.
Image: '/home/blinky/Debug/blinky.elf'.
Command line: 'test 1 2 3' (10 bytes).
Load  13256 bytes at 0x08000000-0x080033C7.
Load    116 bytes at 0x080033C8-0x0800343B.
Load    772 bytes at 0x20000074-0x20000377.
Cortex-M4 r0p0 core initialised.
Unexpected error in object_property_find() at /Host/Work/qemu/gnuarmeclipse-qemu.git/qom/object.c:1007:
qemu-system-gnuarmeclipse: Property '.unnamed-gpio-out[13]' not found
Aborted (core dumped)

ilg-ul, 2016-12-09:
thank you for the bug report.

I'll fix it in the next release.

bestel74, 2016-12-21:
Is there any workaround ?

ilg-ul, 2016-12-21:
use the F4-Discovery board.

ilg-ul, 2016-12-21:
fyi, I just reworked the F429i board, it is again functional, including the reset and user button.

when all boards will be again fully functional, I'll make a new release, hopefully by the end of the year.

bestel74, 2016-12-21:
nice, I gonna test the "-dev" branch so :)
thank you for all your work.

ilg-ul, 2016-12-27:
fixed in 2.8, all boards functional, with LEDs and buttons.

closed 2016-12-27 by ilg-ul