ilg-archived/qemu #36: Question for hardware support support

nix7965, 2017-01-15:
Hi,

My question is that what kind of peripherals or functions does not work except temporary bugs.
I read descriptions in this github and http://gnuarmeclipse.github.io .

I guess this qemu does not support FPU, UART and (maybe) DMA. Other than these two function, are there some components that don't support?

ilg-ul, 2017-01-15:
> I guess this qemu does not support FPU, UART and (maybe) DMA.

not yet, but planned for future versions

> are there some components that don't support?

the purpose of the current version was to run blinky projects, i.e. LEDs and buttons.

the supported peripherals are:

https://github.com/gnuarmeclipse/qemu/tree/gnuarmeclipse-dev/hw/cortexm/stm32

nix7965, 2017-01-15:
In the future, I plan to run some relatively large firmware on this qemu (not now though, and I cannot mention my target firmware). 
Since I have 'almost zero' background on qemu and embedded systems, I want to know which components are not supported and try to avoid using related components. 

So, I also saw some firmwares using some strange component that I don't know. (e.g., SPI and I2C) 
Are they work on it?

ilg-ul, 2017-01-15:
> I have 'almost zero' background on qemu and embedded systems

then I suggest you first get some experience on embedded system and then start digging into the qemu details.

did you run the tutorial from http://gnuarmeclipse.github.io/tutorials/blinky-arm/ ?

nix7965, 2017-01-15:
Yes I did it.

ilg-ul, 2017-01-15:
> Yes I did it.

great. in addition of the LEDs, the latest version also supports the user button, but the tutorial does not uses it.

that's about all you can currently do with qemu.

nix7965, 2017-01-15:
Well... 
I have one more stupid question. So, do some components, such SPI and I2C, work? With tutorial codes, I don't know due to no background.

ilg-ul, 2017-01-15:
> do some components, such SPI and I2C, work?

not yet. the problem is not the mcu peripheral itself, but what is connected via SPI or I2C.

ilg-ul, 2017-01-15:
if you have further questions, please use the project forum

closed 2017-01-15 by ilg-ul