ilg-archived/qemu #39: qemu crashes with error code -1073741819

daveloyall, 2017-02-01:
Hello.

I am following the 'blinky' tutorial on the github.io site.  When I try to debug in qemu, it crashes with error code (return value) -1073741819.  There is no text in the debugging console.  I also get an 'application has stopped working' dialog: 

[image]

So, I went to the command prompt and guessed how to invoke qemu.  There is console output now:

    d:\bin\GNU ARM Eclipse\QEMU\2.8.0-201612271623-dev\bin>qemu-system-gnuarmeclipse
    .exe -d exec --board MCBSTM32F400 --mcu STM32F407IG --verbose --image d:\workspa
    ce.neon\blinky\Debug\blinky.elf
    
    GNU ARM Eclipse 64-bits QEMU v2.8.0 (qemu-system-gnuarmeclipse.exe).
    
    Supported boards:
      Maple                LeafLab Arduino-style STM32 microcontroller board (r5)
      NUCLEO-F103RB        ST Nucleo Development Board for STM32 F1 series
      NUCLEO-F411RE        ST Nucleo Development Board for STM32 F4 series
      NetduinoGo           Netduino GoBus Development Board with STM32F4
      NetduinoPlus2        Netduino Development Board with STM32F4
      OLIMEXINO-STM32      Olimex Maple (Arduino-like) Development Board
      STM32-E407           Olimex Development Board for STM32F407ZGT6
      STM32-H103           Olimex Header Board for STM32F103RBT6
      STM32-P103           Olimex Prototype Board for STM32F103RBT6
      STM32-P107           Olimex Prototype Board for STM32F107VCT6
      STM32F0-Discovery    ST Discovery kit for STM32F051 line
      STM32F4-Discovery    ST Discovery kit for STM32F407/417 lines
      STM32F429I-Discovery ST Discovery kit for STM32F429/439 lines
      generic              Generic Cortex-M board; use -mcu to define the device
    

But, I get the same crash...  `%ERRORLEVEL%` reports the same code.

I tried some `-d` options to get more information, but none that I tried produced output.  I'm not familiar with qemu.

What trouble shooting or log gathering steps would you recommend?

Thanks,
--Dave

ilg-ul, 2017-02-01:
thank you for the bug report, I'll fix the crash in the next release.

what QEMU is trying to tell you is to read carefully the blinky tutorial, since you used a not supported board name.

ilg-ul, 2019-02-26:
Fixed on 2019-02-26.

The new code will issue "Error: Board '%s' is not supported." and exit(1).

closed 2019-02-26 by daveloyall

ilg-ul, 2019-04-24:
Fixed in 2.8.0-5.