ilg-archived/qemu #40: 'SDL/SDL.h'  file not found during build

nscooling, 2017-03-14:
Following the build procedure at:
http://gnuarmeclipse.github.io/qemu/build-procedure/

result of doing:
$ bash build-scripts/scripts/build-qemu.sh —all

I get;
In file included from /Users/Niall/Work/qemu/gnuarmeclipse-qemu.git/hw/display/gpio-led.c:20:
/Users/Niall/Work/qemu/gnuarmeclipse-qemu.git/include/hw/display/gpio-led.h:28:10: fatal error: 'SDL/SDL.h'
      file not found
#include "SDL/SDL.h"

ilg-ul, 2017-03-14:
I just ran a clean build with `--osx` and everything went as expected.

```
Script "build.sh" completed.
do_host_build_target  completed

f351a66a2d31175916fdbf63db1a95faac74148145e6fec03a5dabea2b98fdc5 ?gnuarmeclipse-qemu-osx-2.8.0-201703141900-dev.pkg
081e2852008572ea25a493c3a14f13c1c17ae816acfa275fc0928c6edc923789 ?gnuarmeclipse-qemu-osx-2.8.0-201703141900-dev.tgz

Script "/Users/ilg/Downloads/build-scripts.git/scripts/build-qemu.sh" completed at Tue Mar 14 21:16:14 EET 2017.
Duration: 16 minutes.
```

> /Users/Niall/Work/qemu/gnuarmeclipse-qemu.git/hw/display/gpio-led.c
> /Users/Niall/Work/qemu/gnuarmeclipse-qemu.git/include/hw/display/gpio-led.h

I'm afraid you have a very old version of the repository, those files are no longer there.

the build procedure assumes a recent repository.

do a `cleanall` (which will remove everything except the download folder) and restart the build.

ilg-ul, 2017-03-14:
I also discovered that the instructions to install the Homebrew tools were outdated.

I updated the http://gnuarmeclipse.github.io/qemu/build-procedure/#prerequisites page.

ilg-ul, 2019-02-26:
The new build scripts fixed this problem.

closed 2019-02-26 by ilg-ul