ilg-archived/qemu #41: How to simulate external hardware with the MC?

AJMansfield, 2017-03-21:
I would like to be able to simulate external hardware attached to the MC, so that I can run unit tests to verify the MC will control its peripherals correctly. The external hardware is not very complicated to emulate and I can easily write my own code to do it, but I do not know how to connect it to the emulated MC.

How can I connect emulated external hardware to the QEMU emulation such that I will be able to run unit tests on the system?

ilg-ul, 2017-03-21:
there is no universal recipe. you can take a look how the LED is implemented, the LED object is connected to the output side of a GPIO, and each time the pin changes status, the LED is updated.

input is a bit more complicated, you either poll or use interrupts. you can take a look how the push button is implemented.

what peripheral would you like to emulate?

AJMansfield, 2017-03-21:
I've looked at the way the code is implemented for the button and LED, although I'm not sure if I understand it correctly. If I'm correct, the LED works by setting up the output pin to fire an interrupt on the CPU, and then using semihosting features to change the UI. If so I'm not sure that would work for my application, because I'm mainly trying to test some realtime signal handling - the MC needs to respond to an input edge and immediately set a series of other pins at precise time intervals, within only a couple hundred nanoseconds of the interrupt. Any extra interrupt code that has to run whenever that stuff happens would change the timings significantly and make the whole test invalid.

There's also a few other things I'm interested in emulating, such as an analog filter network (with the hardware ADC / PWM) and an I2C character LCD, but those aren't as critical to test as the realtime signal handling.

ilg-ul, 2017-03-21:
> the LED works by setting up the output pin to fire an interrupt on the CPU, and then using semihosting features to change the UI

not exactly. gpio output invokes callbacks on the led object, that print the message on the console and, if graphics are on, pushes some messages on the graphical event queue.

> I'm mainly trying to test some realtime signal handling

probably not realistic. qemu has no idea of time. even timers are very inaccurate, using posix alarms.

closed 2019-02-26 by ilg-ul