ilg-archived/qemu #42: Ethernet Support

lnzmst, 2017-03-25:
What is the actual state in support of networking ? I would be interested in writing basic proof-of-concept applications able to send and receive raw ehernet packets (a TCP/IP stack is not required).
Thanks in advance,
Lorenzo.

ilg-ul, 2017-03-25:
> actual state in support of networking

planned, but no date set.

lnzmst, 2017-03-28:
Thank you Liviu.
  Looking forward for it ! I have also found semihosting does not support
file creation
(fopen() failed with File descriptor value too large), is that correct or
am I missing
some set-up or configuration ?
Best Regards,
Lorenzo.

ilg-ul, 2017-03-28:
> semihosting does not support file creation (fopen() failed with File descriptor value too large)

please provide the exact line you used, so I can reproduce it.

and do not reply by email, use the web interface.