ilg-archived/qemu #43: HEAP_INFO semihosting support

dumarjo, 2017-04-20:
HI,
 I try to use the rdimon spec file with the newlib from gcc 6.3.0 (arm embedded) and I have a problem with the SYS_HEAPINFO syscall. The information reported by QEMU is wrong and I think this information must be passed on command line of QEMU. 

Tomas Kleiber have done an excelent report on gcc answer forum here https://answers.launchpad.net/gcc-arm-embedded/+question/249968

Do you think that it's possible to provide stack pointer and heap info to QEMU ?

Regards

Jonathan

ilg-ul, 2017-04-20:
for reference, here is one documentation page for `SYS_HEAPINFO`:

http://www.keil.com/support/man/docs/armcc/armcc_pge1358787054284.htm

we are talking about heap & stack. the only knowledge qemu has about the application, is from the binary image. the stack base can be extracted from the first word, but how do you suggest to extract the other 3 values?

dumarjo, 2017-04-20:
HI,

If we cannot pass parameter on command line to fill those 3 informations, I think we should use (if possible) specific symbol defined by the linker. 

Is that possible ?

Regards
Jonathan

ilg-ul, 2017-04-20:
> pass parameter on command line to fill those 3 informations

this seems very tedious and error prone

> specific symbol defined by the linker

you mean to read the ELF symbol table? I don't know what support provides QEMU for this, but I think it is not a very good idea to rely on the presence of the ELF symbol table, QEMU itself should be perfectly happy even with binary flash images, i.e. without an ELF; for example you can keep a persistent file with the flash image and use it every time QEMU starts, and emulate some kind of bootloader, to modify the flash image and save it when QEMU quits.

any reason for your application relying on `SYS_HEAPINFO`? do you really need this? did you use this in other debugging environments? if so, can you give more details on the use case?

I used semihosting intensively for running unit tests, but apart from getting the command line, returning an exit code and eventually writing a test result file, I don't remember any special need for `SYS_HEAPINFO`.

machim, 2017-04-20:
hi,

I found a problem some time ago in the startup code, please see: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1638340

best regards

ilg-ul, 2017-04-20:
you might be the only one using the newlib startup code.

all projects/frameworks that I know, not to mention CMSIS, provide their own startup code.

dumarjo, 2017-04-20:
Hi,
It's only a matter of portability. I have an option for this. I can use my startup without problem and mimic the crt0.S This is working. 

The thing is the same code is working with a real target. This is why I wonder what's the difference.

Regards

Jonathan

ilg-ul, 2017-04-20:
for experimental projects probably it does not matter, but for real applications I prefer to have full control over the startup code and memory map.

dumarjo, 2017-04-20:
HI,

I think 2 information, heap_base and stack_base,  can be detected.

The stack_base is  equal to the first 4 bytes of the bin file. The second one can be detected with an sbrk(0) call. This should give us the starting of the heap.

The others one are more difficult to get. without QEMU argument, it's not easy. 

We can do simulation for the other 2. 

1- heap_limit  could be stack_base if the SP start after heap_base.
2- stack_limit could be heap_base if the SP start after heap_base.

This scenario is probably the most used one. The other one is the opposit.

1- heap_limit  could be end of RAM  if the SP start before heap_base.
2- stack_limit could be start of RAM if the SP start before heap_base.

This scenario is not fail safe because SP can be between data/bss or any other section.

Those solution aren't perfect, but can be a good starting point. 

Is it something doable ?

Regards

Jonathan

ilg-ul, 2017-04-20:
I still do not have a clear picture why you would need to issue a `SYS_HEAPINFO` call in your semihosted application. could you provide more details?

dumarjo, 2017-04-20:
Hi,
This is only done when linking with rdimon.specs. The crt0.S (_start) call this. This is the only reason. 

Regards
Jonathan

ilg-ul, 2017-04-20:
one more reason to avoid the newlib startup code.

dumarjo, 2017-04-21:
Hi,
I checked the openocd code, and it's look like the response to SYS_HEAPINFO return all 0 on all argument. This tell RDIMON to not change anything already configured. 

This can be an easy fix I think.

Regards
Jonathan

ilg-ul, 2017-04-21:
yes, if the ARM semihosting specs allow this (please check), it can be done.

ilg-ul, 2019-02-26:
I checked the specs and returning 0 is legal. I'll update the code to do this.

ilg-ul, 2019-02-26:
Fixed since 2019-02-26.

The function now returns all 0 values. According to the specs, the meaning is that the system was not able to calculate the real values, and the startup code should use whatever values are appropriate.

ilg-ul, 2019-04-24:
Fixed in 2.8.0-5.

closed 2019-04-24 by ilg-ul