ilg-archived/qemu #44: Incorrect behaviour of BASEPRI register

schnommus, 2017-05-11:
In the ARM architecture documentation for ARMv7m, it states that:

> The processor does not process any exception with a priority value greater than or equal to BASEPRI.

However, this behaviour is not reflected in this version of QEMU, resulting in incorrect behaviour of some RTOSs. I have verified this against real hardware for comparison.

The issue can be fixed in `hw/intc/arm_gic.c`:
```
--- a/hw/intc/arm_gic.c
+++ b/hw/intc/arm_gic.c
@@ -99,7 +99,7 @@ void gic_update(GICState *s)
 #if defined(CONFIG_GNU_ARM_ECLIPSE)
                 int prio = GIC_GET_PRIORITY(irq, cpu);
                 uint32_t basepri = *(s->basepri_ptr);
-                if ((basepri == 0) || (prio <= basepri)) {
+                if ((basepri == 0) || (prio < basepri)) {
                     if (prio < best_prio) {
                         best_prio = prio;
                         best_irq = irq;
```
The previous behaviour was to not process any exception with priority _greater than_ BASEPRI, whereas the correct behaviour should be to not process any exception with priority _greater than or equal to_ BASEPRI.

ilg-ul, 2017-05-11:
patch applied, thank you.

schnommus, 2017-05-11:
fantastic! cheers for the rapid response.

ilg-ul, 2017-05-11:
btw, more recent changes in qemu removed the dependency between nvic and gic.

time permitting I'll update my fork, so please check again when this happens, to be sure the basepri behaviour is correct.

ilg-ul, 2019-02-20:
Notice: the issue was fixed, but the ticked was not closed, to double check it after updating to the latest sources.