ilg-archived/qemu #46: `stderr:  err: JSON SVD file '(null)' not found.`

vjpr, 2017-07-17:
From https://www.element14.com/community/message/225654/l/err-json-svd-file-null-not-found

`-board STM32F4-Discovery` works but `NUCLEO-F411RE` does not.

ilg-ul, 2017-07-17:
thank you, I'll fix it, but it'll take a while.

vjpr, 2017-07-17:
Hmm, so it looks like a white space issue.

`-board<space><space>NUCLEO-F411RE` works.

`-board<space>STM32F4-Discovery` says "STM32F411RE" not found.

I remember some of these spacing issues occurring in the past.

vjpr, 2017-07-17:
Ah wait, no it doesn't seem to work - my tests were messed up. It simply doesn't throw the error if there's an extra space.

vjpr, 2017-07-17:
If I use `--board generic` I get `err: qemu-system-gnuarmeclipse: Creating object of type STM32F411 failed; no such type.`

ilg-ul, 2019-02-20:
I know it took some time, but could you retry the same configuration?

just that this time use an absolute path for the qemu executable.

ilg-ul, 2019-02-26:
I could not reproduce the problem.

If you still encounter the problem with the latest QEMU, please reopen the ticket, with more details.

closed 2019-02-26 by ilg-ul