ilg-archived/qemu #47: Can't build due to invalid certificate.

ryosa, 2017-07-21:
build-qemu.sh aborted due to the server certificate issue. Any workaround?

> Cloning 'https://github.com/gnu-mcu-eclipse/qemu.git'...
Cloning into 'qemu.git'...
remote: Counting objects: 309552, done.
remote: Total 309552 (delta 0), reused 0 (delta 0), pack-reused 309551
Receiving objects: 100% (309552/309552), 108.06 MiB | 1.78 MiB/s, done.
Resolving deltas: 100% (250375/250375), done.
Submodule 'dtc' (git://git.qemu-project.org/dtc.git) registered for path 'dtc'
Cloning into 'dtc'...
fatal: unable to access 'https://git.qemu-project.org/dtc.git/': Issuer certificate is invalid.
Clone of 'git://git.qemu-project.org/dtc.git' into submodule path 'dtc' failed

ilg-ul, 2017-07-21:
it is the first time I see this error. I don't know how to avoid it, the certificate is self signed.

btw, there were some updates to the build scripts recently, I don't know if the qemu script is functional again. 

I'll not be available for the next week, I'll take a look when I'm back.

ilg-ul, 2019-02-26:
The build scripts were updated, this problem is probably not actual.

closed 2019-02-26 by ilg-ul