ilg-archived/qemu #49: Warn when unsupported interrupt is enabled

tikonen, 2017-09-13:
### Description

QEMU should warn user when code tries to enable STM32 TIM2 interrupt. Currently this is silently ignored as the feature is not supported.

### Steps to Reproduce

1. Enable STM32 TIM2 interrupt

**Expected behaviour:**

As TIM2 is not currently supported, QEMU should warn user in the console for using a unsupported feature.

**Actual behaviour:**

QEMU silently ignores the interrupt enable.

### Versions
GNU ARM Eclipse 64-bits QEMU emulator version 2.8.0 (v2.8.0-646-g2c99a25-dirty)

Code used to enable interrupts:
```
#include <stdio.h>

#include "stm32f4xx.h"
#include "stm32f4_discovery.h"

extern void initialise_monitor_handles(void);

static TIM_HandleTypeDef sTimerInstance = {
    .Instance = TIM2
};

#define timerINTERRUPT_FREQUENCY		( ( unsigned short ) 500 )

void InitTimer()
{
	__HAL_RCC_TIM2_CLK_ENABLE();

    sTimerInstance.Init.Prescaler = 1;
	sTimerInstance.Init.CounterMode = TIM_COUNTERMODE_UP;
	sTimerInstance.Init.Period = 16000000 / timerINTERRUPT_FREQUENCY;
	sTimerInstance.Init.ClockDivision = TIM_CLOCKDIVISION_DIV1;
	sTimerInstance.Init.RepetitionCounter = 0;
	HAL_TIM_Base_Init(&sTimerInstance);

	__HAL_TIM_CLEAR_FLAG(&sTimerInstance, TIM_SR_UIF);

	HAL_TIM_Base_Start(&sTimerInstance);
	HAL_TIM_Base_Start_IT(&sTimerInstance);
}

// never gets called
void TIM2_IRQHandler()
{
	HAL_TIM_IRQHandler(&sTimerInstance);
}

void HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef *htim)
{
	// TODO 
}

int main(void)
{
	initialise_monitor_handles(); // semihosting, required for prints to work

	puts("main() - enter");

	HAL_Init();
	InitTimer();
	HAL_NVIC_SetPriority(TIM2_IRQn, 1, 1);
	HAL_NVIC_EnableIRQ(TIM2_IRQn);

	// dummy main loop
	for(;;) {
	}

	return 0;
}
```

ilg-ul, 2019-02-20:
> QEMU silently ignores the interrupt enable.

in fact, except the SysTick, none of the other timers are implemented.

@BrendanSimon plans to contribute support for TIM*, probably the next release will include support for them.

ilg-ul, 2019-08-15:
we'll keep it open, hopefully in a future version it'll be fixed.