ilg-archived/qemu #50: unbound variable error while running build-qemu.sh

tmineno, 2017-09-17:
### Description
While running build-qemu.sh to build win64 target, the script was aborted with "unbound variable" error after setup files were generated.

### Steps to Reproduce

 ```
$ git clone https://github.com/gnu-mcu-eclipse/build-scripts.git /Work/build-scripts.git
$ bash ~/Work/build-scripts.git/scripts/build-qemu.sh preload-images
$ bash ~/Work/build-scripts.git/scripts/build-qemu.sh --win64
```

**log:**
```
Changing owner to non-root Linux user...
/Host/Work/qemu/scripts/build-helper.sh: line 724: WORK_FOLDER_PATH: unbound variable
```

### Versions

* Build script version
https://github.com/gnu-mcu-eclipse/build-scripts/commit/ba4c7e6a4a0459f8f5814cce3e710cb2b5709590

* operating system
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.3 LTS"
NAME="Ubuntu"
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

ilg-ul, 2017-09-17:
oops!

sorry, I changed the scripts for openocd and risc-v gcc and I forgot to update it for qemu too...

ilg-ul, 2019-02-20:
The build scripts were moved to a separate project, this is no longer valid.

closed 2019-02-20 by ilg-ul