ilg-archived/qemu #52: "Could not create renderer: Couldn't find matching render driver"

cheasonxie, 2017-10-11:
### Description

./qemu-system-gnuarmeclipse -board STM32-P103 -mcu STM32F103RB --image stm32test.elf
**Could not create renderer: Couldn't find matching render driver**

no matter use eclipse or command line, i got the same error

BrendanSimon, 2017-12-21:
Yes.  I am seeing the same thing on my Windows 7 VM.

```
/cygdrive/c/Program Files/GNU ARM Eclipse/QEMU/2.8.0-201612271623-dev/bin//cygdrive/c/Program Files/GNU ARM Eclipse/QEMU/2.8.0-201612271623-dev/bin/qemu-system-gnuarmeclipse.exe  --verbose --verbose --board STM32F4-Discovery -d unimp,guest_errors --semihosting-config enable=on,target=native --image semihost-CM4.axf

GNU ARM Eclipse 64-bits QEMU v2.8.0 (C:\Program Files\GNU ARM Eclipse\QEMU\2.8.0-201612271623-dev\bin\qemu-system-gnuarmeclipse.exe).
Board: 'STM32F4-Discovery' (ST Discovery kit for STM32F407/417 lines).
Board picture: 'C:\Program Files\GNU ARM Eclipse\QEMU\2.8.0-201612271623-dev\graphics\STM32F4-Discovery.jpg'.
Device file: 'C:\Program Files\GNU ARM Eclipse\QEMU\2.8.0-201612271623-dev\devices\STM32F40x-qemu.json'.
Device: 'STM32F407VG' (Cortex-M4 r0p0, MPU, 4 NVIC prio bits, 82 IRQs), Flash: 1024 kB, RAM: 128 kB.
Image: 'semihost-CM4.axf'.
Command line: (none).
Load   6176 bytes at 0x08000000-0x0800181F.
Load    616 bytes at 0x08001820-0x08001A87.
Cortex-M4 r0p0 core initialised.
'/machine/mcu/stm32/RCC', address: 0x40023800, size: 0x0400
'/machine/mcu/stm32/FLASH', address: 0x40023C00, size: 0x0400
'/machine/mcu/stm32/PWR', address: 0x40007000, size: 0x0400
'/machine/mcu/stm32/SYSCFG', address: 0x40013800, size: 0x0400
'/machine/mcu/stm32/EXTI', address: 0x40013C00, size: 0x0400
'/machine/mcu/stm32/GPIOA', address: 0x40020000, size: 0x0400
'/machine/mcu/stm32/GPIOB', address: 0x40020400, size: 0x0400
'/machine/mcu/stm32/GPIOC', address: 0x40020800, size: 0x0400
'/machine/mcu/stm32/GPIOD', address: 0x40020C00, size: 0x0400
'/machine/mcu/stm32/GPIOE', address: 0x40021000, size: 0x0400
'/machine/mcu/stm32/GPIOF', address: 0x40021400, size: 0x0400
'/machine/mcu/stm32/GPIOG', address: 0x40021800, size: 0x0400
'/machine/mcu/stm32/GPIOH', address: 0x40021C00, size: 0x0400
'/machine/mcu/stm32/GPIOI', address: 0x40022000, size: 0x0400
'/machine/mcu/stm32/USART1', address: 0x40011000, size: 0x0400
'/machine/mcu/stm32/USART2', address: 0x40004400, size: 0x0400
'/machine/mcu/stm32/USART3', address: 0x40004800, size: 0x0400
'/machine/mcu/stm32/USART6', address: 0x40011400, size: 0x0400
'/peripheral/led:green' 8*10 @(258,218) active high '/machine/mcu/stm32/GPIOD',12
'/peripheral/led:orange' 8*10 @(287,246) active high '/machine/mcu/stm32/GPIOD',13
'/peripheral/led:red' 8*10 @(258,274) active high '/machine/mcu/stm32/GPIOD',14
'/peripheral/led:blue' 8*10 @(230,246) active high '/machine/mcu/stm32/GPIOD',15
'/peripheral/button:reset' 40*40 @(262,324)
'/peripheral/button:user' 40*40 @(262,164) active high '/machine/mcu/stm32/GPIOA',0
Cortex-M4 r0p0 core reset.

Could not create renderer: Couldn't find matching render driver
```

ilg-ul, 2019-02-20:
I remember I had a problem when qemu was started with a relative path.

could you test again with the most recent version, and both with absolute and relative path for the qemu executable?

ilg-ul, 2019-02-26:
@BrendanSimon, does this issue still affect the latest release?

BrendanSimon, 2019-02-26:
I'll refresh my memory and take a look when I can

ilg-ul, 2019-02-26:
what happened to your dns domain? it is no longer visible.

BrendanSimon, 2019-02-27:
Works ok on Mac with relative and absolute paths calling the qemu executable.
Now to try and dig up the Windows 7 VM I was using (fingers crossed).

ilg-ul, 2019-02-27:
yes, this was my surprise too, both relative and absolute paths were ok on macOS.

BrendanSimon, 2019-03-09:
Works ok on Mac with relative and absolute paths calling the qemu executable.
Now to try and dig up the Windows 7 VM I was using (fingers crossed).

Unfortunately no luck.  I'm happy for this to be closed.

ilg-ul, 2019-03-09:
ok, in case you encounter this error, you can re-open it.

closed 2019-03-09 by ilg-ul