ilg-archived/qemu #53: Compile to and run on Raspberry Pi

janoist1, 2017-10-24:
Hi,
I'd like to compile to Raspberry Pi 3 (Arm) platform. According to the docs:

> The latest version of the build script is a single run, multi-platform build, generating the Windows 32, Windows 64, GNU/Linux 32, GNU/Linux 64 and macOS distribution packages at once.

Is it possible to compile and run this version of Qemu on Raspberry Pi? Thanks.

ilg-ul, 2017-10-24:
good question. 

but the main problem is not the build script, but QEMU itself. I do not know if qemu is able to run on your ARM platform. 

anyway, unless you plan to distribute binaries for ARM, I would not use my build script, but a modified version, that runs entirely on the native platform, without using Docker images.

janoist1, 2017-10-24:
Thanks for the quick reply.
I think Qemu can be run on a Raspberry Pi.
It is not my goal to distribute binaries for ARM, I'm just an enthusiastic who's giving an attempt to emulate BaseFlight on a Raspberry Pi. I have not gone through all the docs, please apologise if the question is inappropriate: Can you outline how to emulate it? What will I need? Can I use the `.hex` firmware or need to use the source compiled in a special way? Many thanks.

ilg-ul, 2017-10-24:
> I think Qemu can be run on a Raspberry Pi. ... it is not my goal to distribute binaries for ARM

in this case GNU MCU Eclipse QEMU is of little use for you, please build it from the master repository and use it according to the standard documentation.

closed 2019-02-26 by ilg-ul