ilg-archived/qemu #56: a bug in nvic.c:nvic_readl()

dongmu, 2018-02-06:
The new release 2.8.0-20170301, has a bug in nvic:nvic_readl() function. When dealing with "case 0xd04:RETTOBASE", it cannot set a right value. I read the code of Qemu-2.10.0 and add a similar function "nvic_rettobase" to determine whether the value should be changed and the error disappear.
So is there any plan to migrate to Qemu-2.10 version, as the new version is for 2.8?

I want to implement Ethernet functionality for STM32 into present code base. Do you have any suggestions for that?

Any reply will be appreciated.

ilg-ul, 2018-02-06:
I plan to update qemu during the next few weeks.

as for Ethernet, it would be interresting, but I do not know how complicated it is.

ilg-ul, 2019-02-26:
I checked the new source code and there are many changes, it is too complicated to back port them to this old source base.

I'll leave the issue open and revisit it after upgrading to the new souce base.