ilg-archived/qemu #57: Starting QEMU from console, for unit tests, fails.

ilg-ul, 2018-04-29:
The latest version fails when started from console, without GDB.

BrendanSimon, 2018-09-05:
I download your latest release and I couldn't get the semihosting to work at first, but after some googling and experimenting I finally did manage to get it to work (with STM32F4 Disco).  I had to change the linker script ram address (can't remember off the top of my head what it was exactly.  0x10000000 to 0x20000000 or vice versa.  Something like that anyhow).

I'm guessing I might have been able to do it via the gcc/ld command line too (instead of modifying the default gcc linker script)?

The original problem I reported may not have actually been a bug.  It was an older release and I don't really need to revisit it as the new release does work with the correct settings :)

ilg-ul, 2018-09-05:
Ok, then we'll close this ticket.

If you identify problems with semihosting, or with the linker scripts, please discuss them on the forum; if you are sure they are bugs, and you can describe the procedure to reproduce them, please report them via the proper tracker.

closed 2018-09-05 by ilg-ul