ilg-archived/qemu #58: support for cortex-m4 DWT registers available?

echowwww, 2018-06-15:
### Description

When i tried to use DWT registers to count execution cycles, the result was always 0. So i wonder whether the gnu-mcu-eclipse supports DWT register for cortex m4? 

### Steps to Reproduce

1. enable and initiate DWT
2. runs lines of instructions
3. get cycle from DWT_CYCCNT

**Expected behaviour:** [What you expected to happen]
any value greater than 0
**Actual behaviour:** [What actually happened]
0
### Versions
* QEMU v2.8.0-3
* Eclipse Oxygen.2 

Btw, if DWT is really not implemented yet, how should i count the cycle?
Thanks for any advice and help!

ilg-ul, 2018-06-15:
Counting cycles is somehow tricky on an emulator like QEMU, since the original code is translated into native code. I don't know if it is supported.

I'll check next time I'll update to the latest version, and, if available, I'll try to make the count available.

ilg-ul, 2019-02-26:
DWT (Data Watchpoint and Trace Unit) is present on M3/M4 at 0xE0001000–0xE0001FFF.

The DWT->CYCCNT is at 0xE0001004. 

DWT->CTRL is at 0xE0001000. 
[0] | CYCCNTENA | Enable the CYCCNT counter. If not enabled, the counter does not count and no event is generated for PS sampling or CYCCNTENA. In normal use, the debugger must initialize the CYCCNT counter to 0.

---

I checked the current code, and there is no easy way to get the cycles count.

I'll leave this open and I'll revisit it after updating to the latest sources.