ilg-archived/qemu #60: socket programming throught semihosting

duslabo, 2018-09-11:
Hello,

Device: cortex M4 based
qemu-system-gnuarmeclipse version: 2.8.0-3

1. Is there socket programming available through semihosting? or is it planned? 

2. If socket is not availabe through semihosting, how I can achieve communication in qemu. I use lwip stack in my device.

Thanks

duslabo, 2018-09-20:
My understanding was, there are some system calls (host) supported in qemu. like  file read, write, printing to stdout etc.  reference: https://github.com/gnu-mcu-eclipse/eclipse-qemu-test-projects/blob/master/cm3-fs-shd/system/include/arm/semihosting.h 

Similar way, we can also support linux socket calls? like connect, send and recv ? which is POSIC compliant. 

Currently I am not very sure how to implement this, but if anybody throws an idea I can start the implementation here.

ilg-ul, 2018-09-20:
The semihosting implementation is conform to an ARM document (https://static.docs.arm.com/100863/0200/semihosting.pdf) which does not cover network calls.

duslabo, 2018-09-20:
Thanks for this info, I didn't know that header created based on ARM document, so networking in semihosting as of now. 

I should look for network stack thing.. Thanks @ilg-ul for the information...