ilg-archived/qemu #61: Conditional branch after if-then instruction not working

higaski, 2018-11-27:
Hello

There seems to be an issue with QEMU when debugging if-then condition blocks from the thumb2 instruction sets. The following snippet runs fine during normal execution, but keeps hanging at the conditional branch when debugging. The jump at the branch should only be executed as long as $r0 is lower than $r1. Problem is that once both are equal, the execution is not continued past the branch and the program counter never gets popped.
```
2000407a:   push    {lr}
2000407c:   movs    r0, r6
2000407e:   ldmia   r7!, {r1, r6}
20004080:   push    {r0, r1}
20004082:   str.w   r6, [r7, #-4]!
20004086:   ldr     r6, [sp, #0]
20004088:   pop     {r0, r1}
2000408a:   adds    r0, #1
2000408c:   cmp     r0, r1
2000408e:   itt     lt
20004090:   pushlt  {r0, r1}
20004092:   blt.w   0x20004082      ; unpredictable <IT:lt>  // <-- GDB hangs here
20004096:   pop     {pc}
```

I've captured some trace output for single stepping from the pushlt to the blt.w instruction with the trace arguments unimp, guest_errors, op, int, exec:
[qemu_trace_output.txt](https://github.com/gnu-mcu-eclipse/qemu/files/2620759/qemu_trace_output.txt)

The versions I've been running on are QEMU 2.8.0-3-20180523-0703 and arm-none-eabi-gdb 8.2.

ilg-ul, 2018-11-27:
Sorry to hear that QEMU is no emulating correctly.

Unfortunately I'm not able to fix this, I only pack the original emulator and add the board definitions.

It would be interresting to test the same sequence of code on a recent QEMU, the problem might be already fixed.

higaski, 2018-11-27:
Took me a while but I got QEMU 3.0.0 running and it has the same behavior. Looks like I'm gonna move that issue over to the QEMU repository. :)

higaski, 2018-12-02:
Now reported here:
https://bugs.launchpad.net/qemu/+bug/1806243

For some reason I was not able to reproduce the bug with code running from flash memory...
But lets see if anyone over at QEMU has an idea.

closed 2019-02-20 by ilg-ul