ilg-archived/qemu #63: Which board or device's sram is greater than 516K, how config sram

lujnan, 2019-02-25:
### Prerequisites

In case you encounter problems while using the **GNU MCU Eclipse QEMU**, please consider the following:

* [ ] read the documentation pages carefully
* [ ] check the [Known issues](http://gnuarmeclipse.github.io/support/known-issues/)
* [ ] check the [FAQ](http://gnuarmeclipse.github.io/support/faq/)
* [ ] check the [GNU MCU Eclipse forum](http://gnuarmeclipse.github.io/support/forum/) for similar problems
* [ ] check the GitHub Issues [tracker](https://github.com/gnu-mcu-eclipse/qemu/issues/)

If you still could not find a solution, if you have interesting use cases, if you have custom 
configurations, and generally if you have any experience that you want to share with others, 
use the [GNU MCU Eclipse forum](http://gnuarmeclipse.github.io/support/forum/).

Just [register](http://www.element14.com/community/create-account.jspa) to element14 community and use the [Start a discussion](http://www.element14.com/community/discussion/create.jspa?containerID=2436&containerType=700) link, with the QEMU category.

If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a 
pertinent suggestion how to enhance GNU MCU Eclipse QEMU, continue and register
an issue, but please keep in mind that QEMU is a separate project, and functional changes should normally be addressed to the original project developers.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behaviour:** [What you expected to happen]

**Actual behaviour:** [What actually happened]

### Versions

* QEMU package version
* plug-in version
* Eclipse version
* Java version
* operating system
* toolchain version
* the **detailed and complete list of steps to reproduce the bug** (mandatory)

Please understand that without being able to reproduce the bug we cannot identify your problem.

To be allowed to enter issues, you need first to login to GitHub.

Note: the text should respect the markdown syntax; preview the message before posting and correct if it does not look as expected, especially inline code or other quoted text.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2019-02-25:
where is the bug?

lujnan, 2019-02-25:
qemu-system-gnuarmeclipse --verbose --verbose --board STM32F4-Discovery \
--mcu STM32F407VG -d unimp,guest_errors \
**-global sram-size-kb=512 \**
--nographic --image mytest.elf \
--semihosting-config enable=on,target=native \
--semihosting-cmdline mytest

lujnan, 2019-02-25:
the option '-global sram-size-kb' is not available. how can i run and debug demo on qemu that need more memory?

ilg-ul, 2019-02-25:
The list of supported command line options is: https://gnu-mcu-eclipse.github.io/qemu/options/

Where does the `-global` option come from?

lujnan, 2019-02-25:
**global**
'''
-global driver.property=value
-global driver=driver,property=property,value=value
                set a global default for a driver property
'''
**sram-size-kb**
from mcu.c source file.

ilg-ul, 2019-02-25:
Ah, ok, I'll check what happened to `-global`, and if still legal, I'll fix it and add to the documentation.

lujnan, 2019-02-25:
Hi, @ilg-ul I see 'qemu' has another option
```
-m[emory] [size=]megs[,slots=n,maxmem=size]
                configure guest RAM
                size: initial amount of guest memory
                slots: number of hotplug slots (default: none)
                maxmem: maximum amount of guest memory (default: none)
```

ilg-ul, 2019-02-25:
thank you @lujnan for the hint, qemu has lots of options, some contradictory, I'm not sure which one were available when I added the Cortex-M code.

if you are comfortable with checking the source code, any additional help to identify the best solution will be highly appreciated.

ilg-ul, 2019-02-25:
and a short comment/advice: 

although it is possible in emulators to extend the memory size, this is generally not recommended, since one of the usual requirements is to validate the image intended for emulation by running it on a physical board.

as long as everything works just fine you do not face this, but as soon as something breaks, you'll question/suspect the emulator and probably want to run the image on a physical board.

lujnan, 2019-02-26:
@ilg-ul  Is it possible to create a specific board like the android does. (google's android emulator base on qemu, the board named 'goldfish'.)

ilg-ul, 2019-02-26:
probably it is possible, but the focus of this project is Cortex-M, so for Linux devices you should use the 'standard' QEMU.

lujnan, 2019-02-26:
@ilg-ul   I have resolved my question by using the following sample. I now to close  this issue. Thank you very much! 

```
qemu-system-gnuarmeclipse --verbose --verbose --board generic \
--mcu STM32F429ZI --gdb tcp::1234 -d unimp,guest_errors \
--semihosting-config enable=on \
--semihosting-cmdline blinky
```

ilg-ul, 2019-02-26:
yes, using a board with larger memory is the correct solution.

however, I'll reopen this ticket, since I still need to check why -global failed.

ilg-ul, 2019-02-26:
`-global sram-size-kb=512` is not legal, the correct syntax is `-global driver.property=value`, and does not apply to memory size.

ilg-ul, 2019-02-26:
Fixed in 2019-02-26.

`-memory size=kb` sets the value into the machine object, and the mcu reads it from there.

TODO after publishing release: update the documentation.

ilg-ul, 2019-04-24:
Fixed in 2.8.0-5.

closed 2019-04-24 by ilg-ul