ilg-archived/qemu #64: emulate the zephyr os failed!

duanwujie, 2019-03-21:
### Prerequisites

1. Need config the zephyr os develop env and build zephyr

### Description

emulate the zephyr os failed at qemu-system-gnuarmeclipse

### Steps to Reproduce

1. build the zephyr os using bellow command

```
source zephyr-env.sh
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
export GNUARMEMB_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-8-2018-q4-major
export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk/
mkdir samples/hello_world/build
cd samples/hello_world/build/
cmake -GNinja -DBOARD=stm32f4_disco
nijia
```
    The zephyr.elf will be find at ./samples/hello_world/build/zephyr/zephyr.elf
2. Use bellow command to emulating
```
#!/bin/sh
MCU=STM32F429ZI
BOARD=STM32F429I-Discovery
$QEMU --verbose --verbose \
    --board $BOARD \
    --mcu $MCU \
    --image zephyr.elf \
    -d unimp,guest_errors \
    --nographic \
    --semihosting-config enable=on,target=native \
    --semihosting-cmdline test 1 2 3
```

3. The output error
```
linx@core-build:~/hacking/qemu-mcu$ ./run.sh 

GNU MCU Eclipse 64-bit QEMU v2.8.0 (qemu-system-gnuarmeclipse).
Board: 'STM32F429I-Discovery' (ST Discovery kit for STM32F429/439 lines).
Device file: 'STM32F429x-qemu.json'.
Device: 'STM32F429ZI' (Cortex-M4 r0p0, MPU, 4 NVIC prio bits, 91 IRQs), Flash: 2048 kB, RAM: 128 kB.
Image: 'zephyr.elf'.
Command line: 'test 1 2 3' (10 bytes).
Load  30360 bytes at 0x08000000-0x08007697.
Load    344 bytes at 0x08007698-0x080077EF.
Load   4408 bytes at 0x20000000-0x20001137.
Cortex-M4 r0p0 core initialised.
'/machine/mcu/stm32/RCC', address: 0x40023800, size: 0x0400
'/machine/mcu/stm32/FLASH', address: 0x40023C00, size: 0x0400
'/machine/mcu/stm32/PWR', address: 0x40007000, size: 0x0400
'/machine/mcu/stm32/SYSCFG', address: 0x40013800, size: 0x0400
'/machine/mcu/stm32/EXTI', address: 0x40013C00, size: 0x0400
'/machine/mcu/stm32/GPIOA', address: 0x40020000, size: 0x0400
'/machine/mcu/stm32/GPIOB', address: 0x40020400, size: 0x0400
'/machine/mcu/stm32/GPIOC', address: 0x40020800, size: 0x0400
'/machine/mcu/stm32/GPIOD', address: 0x40020C00, size: 0x0400
'/machine/mcu/stm32/GPIOE', address: 0x40021000, size: 0x0400
'/machine/mcu/stm32/GPIOF', address: 0x40021400, size: 0x0400
'/machine/mcu/stm32/GPIOG', address: 0x40021800, size: 0x0400
'/machine/mcu/stm32/GPIOH', address: 0x40021C00, size: 0x0400
'/machine/mcu/stm32/GPIOI', address: 0x40022000, size: 0x0400
'/machine/mcu/stm32/USART1', address: 0x40011000, size: 0x0400
'/machine/mcu/stm32/USART2', address: 0x40004400, size: 0x0400
'/machine/mcu/stm32/USART3', address: 0x40004800, size: 0x0400
'/machine/mcu/stm32/USART6', address: 0x40011400, size: 0x0400
'/peripheral/led:green' 10*8 @(519,109) active high '/machine/mcu/stm32/GPIOG',13
'/peripheral/led:red' 10*8 @(519,130) active high '/machine/mcu/stm32/GPIOG',14
QEMU 2.8.0 monitor - type 'help' for more information
(qemu) Cortex-M4 r0p0 core reset.

NVIC: SCR and CCR unimplemented
NVIC: SCR and CCR unimplemented
NVIC: fault status registers unimplemented
NVIC: fault status registers unimplemented
NVIC: Bad write offset 0xd94
NVIC: Bad write offset 0xd98
NVIC: Bad write offset 0xd9c
NVIC: Bad write offset 0xda0
NVIC: Bad write offset 0xd98
NVIC: Bad write offset 0xd9c
NVIC: Bad write offset 0xda0
NVIC: Bad write offset 0xd98
NVIC: Bad write offset 0xd9c
NVIC: Bad write offset 0xda0
NVIC: Bad write offset 0xd94

(qemu) 

```

### Versions

* QEMU:   GNU MCU Eclipse 64-bit QEMU v2.8.0 (qemu-system-gnuarmeclipse)
* Debian GNU/Linux 9.6 (stretch)
* zephyr  1.13.99
* zephyr-sdk 0.9.5  
* gcc-arm-none-eabi-8-2018-q4-major  for zephyr

### urls
* zephyr-sdk: https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/download/0.9.5/zephyr-sdk-0.9.5-setup.run

* zephyr: https://github.com/zephyrproject-rtos/zephyr

### Attached File
- zephyr.elf  [z.zip](https://github.com/gnu-mcu-eclipse/qemu/files/2990649/z.zip)

ilg-ul, 2019-03-21:
unfortunatelly the emulation of NVIC is incomplete.

I plan to address this in a future version, but for now try to avoid the missing registers.

duanwujie, 2019-03-21:
I disabled the MPU config in the zephyr  then "Bad write offset" disappeared. Output bellow:
```
QEMU 2.8.0 monitor - type 'help' for more information
(qemu) Cortex-M4 r0p0 core reset.

NVIC: SCR and CCR unimplemented
NVIC: SCR and CCR unimplemented
NVIC: fault status registers unimplemented
NVIC: fault status registers unimplemented

(qemu) 

```

ilg-ul, 2019-03-21:
Ok, try to identify the code using SCR/CCR and disable it too.

Apart from these messages, does the application start?

duanwujie, 2019-03-21:
Haven't started

ilg-ul, 2019-03-21:
Could you place a breakpoint on the start code (the reset vector) and single step to see how far it gets?

duanwujie, 2019-03-21:
Everytime i start the qemu it stop at qmonitor, the option "-s -S" notworking !

ilg-ul, 2019-03-21:
Not like this, run a debug session. For this start QEMU as a GDB server, and direct a GDB client to it.

If you did not do this before, I suggest you follow the GNU MCU Eclipse blinky tutorial and learn how to run debug sessions on QEMU.

duanwujie, 2019-03-21:
It's huang in the uart:
```
(gdb) target remote :1234
Remote debugging using :1234
0x08004ae0 in uart_stm32_poll_out (dev=0x200011a8 <__device_uart_stm32_USART_2>, c=105 'i')
    at /home/duanwujie/hacking/linx-zephyr/drivers/serial/uart_stm32.c:390
390		while (!LL_USART_IsActiveFlag_TXE(UartInstance))
```

ilg-ul, 2019-03-21:
To output trace messages, use semihosting calls, not the uart.

ilg-ul, 2019-03-21:
If I remember right, QEMU also implements ARM ITM, it is even faster than semihosting.

ilg-ul, 2019-03-21:
@BrendanSimon, here is another example why the curent model of dealing with peripherals is not ideal; peripherals that are not implemented properly should not be there at all; in this particular case the uart should be fixed, but all other not implemented peripherals should display an error when accessed, to know that they are not implemented.

duanwujie, 2019-03-22:
I am debugging in eclipse-mcu, it works fine,except the output message。

ilg-ul, 2019-03-22:
Redirect it to semihosting or ITM and it'll be fine. For examples how to do it, see the projects generated by the STM32 project templates.

duanwujie, 2019-03-22:
After redirect to semihosting it's worked fine, now  i plan to fixed the uart side effect to roburst qemu-mcu.
ths.

ilg-ul, 2019-03-22:
> i plan to fixed the uart side effect to roburst qemu-mcu.

I'm not sure I understand this, but if you manage to fix QEMU, please send PRs and I'll analyse them.

I recently updated the QEMU build scripts and it is now easier to build it for development purposes.

duanwujie, 2019-03-25:
I want to add the support for the nucleo-f767zi, no i have get STM32F7x.json form svd file。According the qemu-mcu code , i just need to add the board class for  nucleo-f767zi ! Do you have any suggestion to port this ?

duanwujie, 2019-03-25:
Is there any document for developer ?

ilg-ul, 2019-03-25:
> i just need to add the board class for nucleo-f767zi

well, you need to add the board class, with picture, leds & buttons positions, but you also must be sure the device class is implemented properly, and then add functionality for the mandatory devices.

unfortunatelly there is no documentation, you need to check the source code. the reference is the stm32f4-discovery board.

there is some rework planned for qemu, to further automate adding support for new devices, but it might take some time.

duanwujie, 2019-04-03:
there have problem for emulating nucleo-f767zi board. The system clock is slowlly.

ilg-ul, 2019-04-03:
can you estimate how much slower?

duanwujie, 2019-04-04:
I find the problem. In the RCC the nucleo-f767zi use hsi as clock source cpu_freq_hz = 16000000

ilg-ul, 2019-04-04:
Does QEMU run at the same speed as the physical board now?

duanwujie, 2019-04-04:
I just change  the cpu_freq_hz to 400MHz. No it's worked in 400Mhz and here is the screenshot.
[image]

duanwujie, 2019-04-04:
But in the 400MHz will huang up! here's output
```
main-loop: WARNING: I/O thread spun for 1000 iterations
```

ilg-ul, 2019-04-04:
I have difficulties to follow you. Please run on QEMU only applications that you previously tested on the physical board and are confirmed to be functional.

If you identify functional differences, explain them clearly such that I can reproduce them.