ilg-archived/qemu #66: Is bug of STM32F429x-qemu.json

duanwujie, 2019-04-10:
I find the usart's status register's reset value is different from datasheet. snippet bellow.
```
"registers": [
					{
						"name": "SR",
						"displayName": "SR",
						"description": "Status register",
						"addressOffset": "0x0",
						"size": "0x20",
						"resetValue": "0x00C00000",
```

The reset value is 0x00C00000
But in the datasheet the reset value is 0x0000 00C0 

Is this big endian question ?

datasheet : RM0090 Reference manual STM32F405/415, STM32F407/417, STM32F427/437 and
STM32F429/439 advanced Arm®-based 32-bit MCUs

ilg-ul, 2019-04-10:
> Is this big endian question ?

probably.

I'll check it.

ilg-ul, 2019-04-10:
The `*-qemu.json` files are automatically generated from the original Keil `*.svd`. In those files the value is `<resetValue>0x00C00000</resetValue>`.

The error probably comes from the STM manuals, which, in previous versions, listed `Reset value: 0x00C0 0000`.  However, the recent version, RM0090 Rev 18, lists the correct value.

For the moment there is no proper fix, the old xcdl tool used to patch the xsvd files needs to be updated to allow patching the reset value field.

As a workaround I'll manually edit the files.

Anyway, please note that USART support is not yet functional.

ilg-ul, 2019-04-24:
Fixed in 2.8.0-5.

closed 2019-04-24 by ilg-ul