ilg-archived/qemu #68: Qemu failed, xpack GNU MCU Eclipse, OpenSuse Leap 15: "Unable to initialize SDL: No available video device"

user4757, 2019-05-08:
### Description

I wanted to start with STM32 Microcontroller und tried the "blinky" example with GNU MCU Eclipse as destribed here:
- Tutorial: Create a Blinky ARM test project, https://gnu-mcu-eclipse.github.io/tutorials/blinky-arm/

All works fine until "Run the test with QEMU". **Qemu did not start.**

The only output in Eclipse Console was:
"GNU MCU Eclipse 64-bit QEMU v2.8.0-5 (qemu-system-gnuarmeclipse)."
and over there was a line 
"<terminated> Test blinky Debug [GDB Qemu Debugging] qemu-system-gnuarmeclipse"
There was **no error message**!

I didn't find the called "command line" in Eclipse, but after some time with google:
/home/user/opt/xPacks/@gnu-mcu-eclipse/qemu/2.8.0-5.1/.content/bin/qemu-system-gnuarmeclipse --verbose --verbose --board STM32F4-Discovery --mcu STM32F407VG --gdb tcp::1234 -d unimp,guest_errors --semihosting-config enable=on,target=native --semihosting-cmdline blinky
failed with error message:
GNU MCU Eclipse 64-bit QEMU v2.8.0-5 (qemu-system-gnuarmeclipse).
**Unable to initialize SDL:  No available video device**

But there is a video device!
After renaming opt/xPacks/@gnu-mcu-eclipse/qemu/2.8.0-5.1/.content/bin/libSDL2-2.0.so.0 (to _libSDL2-2.0.so.0) qemu used the system wide sdl library (OpenSuse Leap 15) and **qemu works**!

I think there is a bug with sdl library coming with qemu xpacks.

### Steps to Reproduce

I installed GNU MCU Exlipse and other software as recommendet in the "blinky" example:
https://gnu-mcu-eclipse.github.io/install/

I used xpm to install gcc, build tools, openocd and qemu.

### Versions

* GNU MCU Eclipse 64-bit QEMU v2.8.0-5 (qemu-system-gnuarmeclipse)
* Eclipse 2018-12 (4.10.0)
* OS OpenSuse Leap 15

ilg-ul, 2019-05-08:
You are right, the sdl2 library for linux has a problem, it is mentioned in the Known problems.

Most probably, due to a mistake in the new build environment, it was compiled without x11 support.

It is on my TODO list, hopefully will be solved during the next days.

As a workaround, you can use the system library.

ilg-ul, 2019-05-17:
Fixed in 2.8.0-6 20190517.

closed 2019-05-17 by ilg-ul