ilg-archived/qemu #69: Blinky on QEMU with STM32F1 fails

cooked, 2019-05-29:
### Description

The default blinky project used to [test the correct functioning of QEMU](https://gnu-mcu-eclipse.github.io/tutorials/blinky-arm/) does not work for the STM32F1 targets. 

### Steps to Reproduce
1. create new project from File>New>Others
2. selected C++ Project and the STM32F10x Executable template with the ARM Cross GCC toolchain
3. select default target options as below
[image]
4. select folders settings default,
5. create Debug/Release configurations default GNU MCU Eclipse ARM Embedded GCC (arm-none-eabi-gcc) toolchain and complete
6. built the project, Debug configuration
7. assigned the NUCLEO STM32F103RB device (or equivalent NUCLEO board) to the project
[image]
8. create and launch debug config
[image]

**Expected behaviour:** 
QEMU to run the blinky project succesfully

**Actual behaviour:**
QEMU throws the following error

[image]

### Versions

* QEMU package version :  2.8.0-6
* QEMU plug-in version:  3.3.3.201901011632
* Eclipse version: 4.11.0
* Java version: OpenJDK 1.8.0_212 64 bit
* operating system: Linux Debian 64bit
* toolchain version: 8.2.1-1.7-20190524-0606

cooked, 2019-05-29:
Also the STM32F0x seems to fail in a similar way.

ilg-ul, 2019-05-29:
thank you, I'll investigate.

randre70, 2019-08-14:
Also applies to Windows version. Tried several configuration options of cpu and qemu to no avail.
[image]
[image]

ilg-ul, 2019-08-14:
could you try the latest version? it is available from 

https://github.com/xpack-dev-tools/qemu-arm-xpack/releases

randre70, 2019-08-14:
Same error
[image]
[image]

ilg-ul, 2019-08-14:
thank you, I entered a new ticket, I'll investigate and try to fix it in the next release.

ilg-ul, 2019-11-04:
Please check 2.8.0-8 and confirm if the bug was fixed.

closed 2019-11-04 by ilg-ul